enesbcs/rpieasy #1: Great Concept

happytm, 2018-12-30:
Thank you for starting great concept. I am user of great Espeasy Software on esp8266 for long time. RPIeasy could be very useful if there is I2C bridge with arduino pro mini thereby getting rid of some limitations.

Thanks for great start.

enesbcs, 2018-12-30:
Where can i find instructions about creating and programming the Pro Mini?

happytm, 2018-12-30:
As you can see I am using Pro Mini for Mains current measurement and AC dimmer control which are impossible using only RPI or ESP8266.
That is the importance of including Pro Mini. Only requirement is communication going between RPI and Pro Mini via I2C bus.

Thanks

enesbcs, 2018-12-30:
I found this on ebay, according to its description it is an AT Mega328 3.3V compatible with Pro Mini, i can order this if it will be sufficient to test:
https://www.ebay.com/itm/Mini-ATMEAG328-3-3V-8Mhz-Replace-ATmega128-for-Arduino-Pro-Mini-Compatible/173360751835
Otherwise I2C is not a big deal, but the ESPEasy plugin that handles the Pro Mini looks like a little complicated to me, it will not be easy to port it. Sending commands to the Pro Mini may be implemented, but for the reverse way (from ProMini to RPIEasy) currently i have no clue how to store/display data...

happytm, 2018-12-30:
I followed user Costo's post on ESPeasy forum to understand how to read values from Pro Mini into ESPeasy. Basically he set up devices in ESPeasy on devices tab to read value from Pro Mini.Link is below

https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=627&p=6385&hilit=pme#p6385

enesbcs, 2018-12-31:
So basically we can use Dummy devices to hold values and create one (or more) ProMini devices, in which to be given the destination task and value number(s) (max 10 row per dev?) with type (analog/digital)? I've done something similar (value rerouting) with battery value watching, which can be setted in Advanced settings. (i am using an ADS1015 for this goal now)

The "compatible" device i mentioned in the previous post is right?

happytm, 2018-12-31:
You will also need FTDI adapter to upload the sketch to Pro Mini like below

https://www.ebay.com/itm/FT232RL-FTDI-Serials-Adapter-Module-Mini-Port-f-USB-to-TTL-3-3V-5-5V-BBC/142727421107?hash=item213b379cb3:g:EKMAAOSwinVZtzmB:rk:1:pf:0

Thanks

enesbcs, 2018-12-31:
Thanks, but I already have a CP2102 (for flashing Sonoff devices) i think it will work for uploading, and ordered a ProMini for testing.

enesbcs, 2019-01-15:
I was reading and tried to interpret the PME code. But i found a possible issue: It's default address 0x7f is outside of the valid I2C range. (0x08-0x77)
https://www.totalphase.com/support/articles/200349176-7-bit-8-bit-and-10-bit-I2C-Slave-Addressing
I doubt that RPI smbus module will handle it.

happytm, 2019-01-15:
Is it not possible to define different address within the range (0x008-0x77) in PME plugin  and match same newly defined address in pro mini extender code on arduino ?

happytm, 2019-01-17:
Thank you for start on PME plugin. Can't wait to test once it is implemented.

enesbcs, 2019-01-17:
Experimental version added at https://github.com/enesbcs/rpieasy/commit/b0ca9497a4b213ce7ee0de1cea7d033fd4d35039

happytm, 2019-01-18:
I started testing this plugin and following is my feedback -

- I tested both standard address 0x3f & non standard address 0x7f and both seems to communicate with 
 pro mini so that was very good start.

- I then attached joystick to analog pin 6 & 7 & switch part of joystick to digital pin 9 of pro mini and set up 3 different devices on device page to read value from each of 3 pins it send values but they are all same value and not different value.It should be different value for each of 3 pins. following is log - 

When joystick down -

22:27:46 Event: PMEanalog6#Value=0.0
22:27:46 Event: PMEanalog7#Value=0.0
22:27:47 Event: PMEdigital9#Value=0.0

When joystick in middle -

22:27:50 Event: PMEdigital9#Value=255.0
22:27:50 Event: PMEanalog6#Value=511.0
22:27:50 Event: PMEanalog7#Value=511.0

When joystick up -

22:27:55 Event: PMEanalog6#Value=1023.0
22:27:55 Event: PMEanalog7#Value=1023.0
22:27:56 Event: PMEdigital9#Value=255.0

22:28:00 Event: Clock#Time=Thu,22:28

- When switch pressed on joystick for few seconds -

22:31:52 Event: PMEdigital9#Value=149.0
22:31:52 Event: PMEanalog6#Value=405.0
22:31:52 Event: PMEanalog7#Value=405.0
22:31:53 Event: PMEdigital9#Value=162.0
22:31:53 Event: PMEanalog6#Value=418.0
22:31:53 Event: PMEanalog7#Value=418.0
22:31:54 Event: PMEdigital9#Value=160.0
22:31:54 Event: PMEanalog7#Value=416.0
22:31:54 Event: PMEanalog6#Value=416.0
22:31:55 Event: PMEdigital9#Value=158.0
22:31:55 Event: PMEanalog6#Value=414.0
22:31:55 Event: PMEanalog7#Value=414.0
22:31:56 Event: PMEdigital9#Value=158.0
22:31:56 Event: PMEanalog7#Value=414.0
22:31:56 Event: PMEanalog6#Value=414.0
22:31:57 Event: PMEdigital9#Value=148.0
22:31:57 Event: PMEanalog6#Value=404.0
22:31:57 Event: PMEanalog7#Value=404.0
22:31:58 Event: PMEdigital9#Value=110.0
22:31:58 Event: PMEanalog6#Value=366.0
22:31:58 Event: PMEanalog7#Value=366.0
22:31:59 Event: PMEdigital9#Value=138.0
22:31:59 Event: PMEanalog6#Value=138.0
22:31:59 Event: PMEanalog7#Value=138.0
22:32:00 Event: PMEdigital9#Value=99.0
22:32:00 Event: PMEanalog6#Value=355.0
22:32:00 Event: PMEanalog7#Value=355.0
22:32:00 Event: Clock#Time=Thu,22:32
22:32:02 Event: PMEdigital9#Value=99.0
22:32:02 Event: PMEanalog6#Value=355.0
22:32:02 Event: PMEanalog7#Value=355.0
22:32:03 Event: PMEdigital9#Value=101.0
22:32:03 Event: PMEanalog6#Value=357.0
22:32:03 Event: PMEanalog7#Value=357.0
22:32:04 Event: PMEdigital9#Value=101.0
22:32:04 Event: PMEanalog6#Value=357.0
22:32:04 Event: PMEanalog7#Value=357.0
22:32:05 Event: PMEdigital9#Value=106.0
22:32:05 Event: PMEanalog6#Value=362.0
22:32:05 Event: PMEanalog7#Value=362.0
22:32:06 Event: PMEdigital9#Value=111.0
22:32:06 Event: PMEanalog6#Value=367.0
22:32:06 Event: PMEanalog7#Value=367.0
22:32:07 Event: PMEdigital9#Value=120.0
22:32:07 Event: PMEanalog6#Value=376.0
22:32:07 Event: PMEanalog7#Value=376.0
22:32:08 Event: PMEdigital9#Value=120.0
22:32:08 Event: PMEanalog6#Value=376.0
22:32:08 Event: PMEanalog7#Value=376.0
22:32:09 Event: PMEdigital9#Value=120.0
22:32:09 Event: PMEanalog6#Value=376.0
22:32:09 Event: PMEanalog7#Value=376.0
22:32:10 Event: PMEdigital9#Value=127.0
22:32:10 Event: PMEanalog6#Value=383.0
22:32:10 Event: PMEanalog7#Value=383.0
22:32:11 Event: PMEdigital9#Value=127.0
22:32:11 Event: PMEanalog6#Value=383.0
22:32:11 Event: PMEanalog7#Value=383.0
22:32:12 Event: PMEdigital9#Value=127.0
22:32:12 Event: PMEanalog6#Value=383.0
22:32:12 Event: PMEanalog7#Value=383.0
22:32:13 Event: PMEdigital9#Value=126.0
22:32:13 Event: PMEanalog6#Value=382.0
22:32:13 Event: PMEanalog7#Value=382.0
22:32:14 Event: PMEdigital9#Value=127.0
22:32:14 Event: PMEanalog6#Value=383.0
22:32:14 Event: PMEanalog7#Value=383.0
22:32:15 Event: PMEdigital9#Value=126.0
22:32:15 Event: PMEanalog6#Value=382.0
22:32:15 Event: PMEanalog7#Value=382.0
22:32:16 Event: PMEdigital9#Value=126.0
22:32:16 Event: PMEanalog6#Value=382.0
22:32:16 Event: PMEanalog7#Value=382.0
22:32:17 Event: PMEdigital9#Value=231.0
22:32:17 Event: PMEanalog6#Value=231.0
22:32:17 Event: PMEanalog7#Value=231.0

If I move joystick sideways on both ends it seems there is no effect and values in log remain same as if joystick is middle position.

For output side if at least 1 device is enabled (for reading pin value from pro mini ) then both extpwm & extgpio works fine and turn pin high or low when I issue extpwm or extgpio command from advanced tab of rpieasy. But in original espeasy there was no need to set up pme plugin on device page in order to enable extgpio and extpwm command. Output side worked automatically  without setting up anything on device page for pme plugin. 

It is very good start and I appreciate your efforts.

Thanks

happytm, 2019-01-18:
On further testing I realised that I was using 1 second interval on device page for all 3 pins so i changed interval to 2 seconds on 2 analog pins ( analog 6 & analog 7 ) and interval of 10 seconds on digital pin 9 and then it works reliably as it should. sorry for confusion. Is it possible to change this behavior so that when switch is pressed or released on pin 9 it send data instantly to rpi instead of waiting for 10 seconds ?

enesbcs, 2019-01-18:
Thanks for testing.
In RPIEasy, it will sure commands will remain at their specific plugins, so you have activate "Switch" plugin to use gpio and MCP plugin to use extgpio.. for the same reason you have to activate USBRelay plugin to use usbrelay command, and playaudio
The reason is simple: RPIEasy can be run on machines, that do not have GPIO support, for example i running it at my PC also with Ubuntu. And i think every plugin has it's specific task, for good reason.

I am not familiar with the Arduino side of the ProMini, pin number 9 is just forwarded as it written in the text box to the ProMini.
As i see the promini code is not capable of sending any data without receiving a read command, so i do not know, how to achieve instantous feedbacks from gpio changes. Is that function works on ESPEasy plugin? Maybe i can send several read commands per second, for ex 10 times per second. Or a new function has to be coded to the promini, that registers attachInterrupt() handler with the selected port and sends data accordingly.

happytm, 2019-01-18:
Thank you for your clarification on switch plugin. I will have to test more to see if there can be something done on pro mini code.

enesbcs, 2019-01-19:
@happytm: 
Could you try the new plugin? https://github.com/enesbcs/rpieasy/commit/7373154e620b68eca673402db3e90e57183bf396
Please delete PME devices, than recreate, and set Interval to 0. In this mode plugin asks the selected pin status ten times per second, and if state is changed, than event is fired. Warning: In this mode, i measured approximately +4% CPU usage/pin! So only use, if it is crucial. I tested 1 second interval with two pin and i do not found any problem, and cpu usage is minimal in this mode. (yet)

happytm, 2019-01-20:
I tried new plugin -

- digital pin connected to switch now works instantly without any problem.
- analog pins connected to joystick and interval set at 0 in plugin does not respond in log at all when changing joystick position. So I tried different interval from 1 second to 5 seconds and it respond with joystick activity but it shows exact same value for both.Before this commit it was working ok and showing separate values from 2 pins with 2 second interval in plugin.

enesbcs, 2019-01-20:
I guess that analog readings takes more time at the ProMini side, and i have to set delay back to 0.01s.  (if data not arrived or arrived a lot of zeros, than it is not a changing and with Interval 0 only the changes logged) 
On the other hand, digital reads are almost instantous so i left it at 0.001s. I doubt that analog reading is possible with this device at a faster speed. Is it faster with ESP8266?
Updated on github.

happytm, 2019-01-21:
It did not change duplicate values problem. I tried sending data from pro mini in different ways but no success.I created separate  integers in pro mini for each pin and sent 7 integers to rpieasy but somehow with every 2 second interval (which I set in plugin ) value becomes same even though when I print those integers values in serial monitor on pro mini it shows separate values for each analog pin correctly.

enesbcs, 2019-01-22:
Seems really odd, tried to use a semaphore to not mix results.
RPIEasy uses threading by design to run plugin_read commands to speed things up, so it can happen that an analogread command interrupts an other one's response. (ESPEasy is single threaded so it is not a problem there) 
https://github.com/enesbcs/rpieasy/commit/dce3e3c70f0edec163c1f3f2901072ad0d06e3f7
If latest commit not helps, than i have to extend the arduino sketch to include pin number in the response packet, this is the only way we can sure that results do not mix anymore.

happytm, 2019-01-23:
Whatever you did with this commit it seems duplicate values from pro mini no longer happening. I used joystick to test (using 2 seconds interval in plugin for both pins) and it brings separate values from both pins based on real activity of joystick. Thank you.

enesbcs, 2019-01-23:
you are welcome. in theory 1 seconds interval might work also.
if this operation is adequate, can this issue be closed?

happytm, 2019-01-23:
Spoke too soon. it seems like it behaves like before after few minutes.But if I reset pro mini it works properly for few minitues again and ssame pattern again. the log is below with comments:

01:07:38 Event: analog6#Value=511.0 // problem starts
01:07:39 Event: analog7#Value=511.0
01:07:40 Event: analog6#Value=511.0
01:07:41 Event: analog7#Value=511.0
01:07:42 Event: analog6#Value=511.0
01:07:43 Event: analog7#Value=511.0
01:07:44 Event: analog6#Value=511.0
01:07:45 Event: analog7#Value=511.0
01:07:46 Event: analog6#Value=511.0
01:07:47 Event: analog7#Value=511.0
01:07:48 Event: analog6#Value=511.0
01:07:49 Event: analog7#Value=511.0
01:07:50 Event: analog6#Value=511.0
01:07:51 Event: analog7#Value=511.0
01:07:52 Event: analog6#Value=511.0
01:07:53 [Errno 121] Remote I/O error     // pro mini reset by me manually
01:07:54 [Errno 121] Remote I/O error
01:07:55 [Errno 121] Remote I/O error
01:07:56 Event: analog6#Value=507.0     // from here works fine until automatic reset
01:07:57 Event: analog7#Value=511.0
01:07:58 Event: analog6#Value=507.0
01:07:59 Event: analog7#Value=511.0
01:08:00 Event: analog6#Value=507.0
01:08:00 Event: Clock#Time=Wed,01:08
01:08:01 Event: analog7#Value=511.0
01:08:02 Event: analog6#Value=507.0
01:08:03 Event: analog7#Value=511.0
01:08:04 Event: analog6#Value=507.0
01:08:05 Event: analog7#Value=511.0
01:08:06 Event: analog6#Value=507.0
01:08:07 Event: analog7#Value=511.0
01:08:09 Event: analog6#Value=507.0
01:08:10 Event: analog7#Value=511.0
01:08:11 Event: analog6#Value=507.0
01:08:12 Event: analog7#Value=511.0
01:08:13 Event: analog6#Value=507.0
01:08:14 Event: analog7#Value=511.0
01:08:15 Event: analog6#Value=507.0
01:08:16 Event: analog7#Value=511.0
01:08:17 Event: analog6#Value=507.0
01:08:18 Event: analog7#Value=511.0
01:08:19 [Errno 121] Remote I/O error  //automatic reset 
01:08:20 Event: analog7#Value=511.0  // problem starts again
01:08:21 Event: analog6#Value=511.0
01:08:22 Event: analog7#Value=511.0
01:08:23 Event: analog6#Value=511.0

enesbcs, 2019-01-23:
I found this hint: "if the Master runs too fast and sends another Wire.beginTransmission() before the last read is complete, this can lead to communications bind up as the prior event onReceive() did not complete its work and possibly corrupting the current transmission in progress."
I have to rethink I2C usage in RPIEasy. Other plugins are not so affected , but with multiple PME device started with several thread paralell needs a more exact timing and maybe some global I2C handler. Work in progress.

enesbcs, 2019-01-23:
Please try commit https://github.com/enesbcs/rpieasy/commit/36f9a8724fe0079fa46931a4b6b8726b737be783
I2C queueing implemented to avoid value mixing and i2c grabbing by one plugin. Runned for several minutes without errors, but needs further testing. If ProMini suddenly stops responding, than try to modify line 79 in rpiwire.py from:
`time.sleep(0.02)`
to 
`time.sleep(0.08)`
than remove power from it and reconnect. May be a timing issue, but can also be a buffer overflow error at the Arduino sketch of the ProMini. (I've got a package full of 0xFF bytes from the ProMini right before the same values began to arrive)

happytm, 2019-01-24:
I tested it and it is working better than before.It lasted 1 hour this time before automatic reset.I think you can close the issue at this time and in future you can try it yourself or some one else with promini and joystick.

enesbcs, 2019-01-24:
Thank you very much, your package arrived to me today, and it contains ProMini and joystick also (among many other stuff), so i can do more tests. :) 
I am not sure if it is a reset on the ProMini, as hardware reset solves the problem, instead a lock-up.. which can be solved with a Watchdog i think. 
Did you use the joystick with direct connection with a 3V ProMini or it needs a Level Shifter between the two?

happytm, 2019-01-24:
I used joystick directly because pro mini i always use is 3.3v version (8mhz) the one I sent you is 3.3v so no need to have level shifter. The programming board also is set on 3.3v via jumper. I do not populate 6 pins on edge of pro mini to program it but just use holes and hold it tight with ftdi adapter while loading the program and it works. For my tests I used analog pins A6 & A7 for joystick.

enesbcs, 2019-01-24:
Thanks! I rewrite the MiniProExtender sketch, relocated the slow analogread() function from the ISR into the loop, i will test it with the joystick and let you know the results.

happytm, 2019-01-30:
Oh that's a good news. 
Thank you.

enesbcs, 2019-01-30:
Could you send the sketch you are using, without the unnecessary portions/comments? The one that you included in the first post, has some errors with comments and something is missing, as number of "}" is not ok. The section that calculates ESPrms is commented out, isn't it necessary?

happytm, 2019-01-31:
I am sorry I am not a programmer so I must have messed up somewhere. but following is user Costo's original sketch I am trying to use for promini:

/********************************************************************************************************************\
 * Arduino project "ESP Easy" Copyright www.esp8266.nu
 *
 * This program is free software: you can redistribute it and/or modify it under the terms of the GNU 
 * General Public License as published by the Free Software Foundation, either version 3 of the License, 
 * or (at your option) any later version.
 * This program is distributed in the hope that it will be useful, but WITHOUT ANY WARRANTY; 
 * without even the implied warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
 * See the GNU General Public License for more details.
 * You received a copy of the GNU General Public License along with this program in file 'License.txt'.
 *
 * IDE download    : https://www.arduino.cc/en/Main/Software
 * ESP8266 Package : https://github.com/esp8266/Arduino
 *
 * Source Code     : https://sourceforge.net/projects/espeasy/
 * Support         : http://www.esp8266.nu
 * Discussion      : http://www.esp8266.nu/forum/
 *
 * Additional information about licensing can be found at : http://www.gnu.org/licenses
\******************************************************************************************************************/
// This file is to be loaded onto an Arduino Pro Mini so it will act as a simple IO extender to the ESP module.
// Communication between ESP and Arduino is using the I2C bus, so only two wires needed.
// It is possible to run the Pro Mini on 3V3, although the 16MHz versions do not officially support this at 16MHzl 
// By working on 3.3volt  you can skip levelconverters on I2C, but no guarantee that it will work stable.
// Arduino Mini Pro uses A4 and A5 for I2C bus. ESP I2C can be configured but they are on GPIO-4 and GPIO-5 by default.

#include <Wire.h>

#define I2C_MSG_IN_SIZE    4
#define I2C_MSG_OUT_SIZE   4

#define CMD_DIGITAL_WRITE  1
#define CMD_DIGITAL_READ   2
#define CMD_ANALOG_WRITE   3
#define CMD_ANALOG_READ    4

volatile uint8_t sendBuffer[I2C_MSG_OUT_SIZE];
int ESPrms;                                                          // value to be sent over I2C to ESP

void setup()
{
  Wire.begin(0x7f);
  Wire.onReceive(receiveEvent);
  Wire.onRequest(requestEvent);
}

/* ******************************************************************************************************************
 This part in the loop found here: [url] https://forum.arduino.cc/index.php?topic=179541.0 [/url] tnx to dc42
 constant 75.7576 depends on the sensitivity of the ACS712 module
 Sensitivity Min<Typ<Max mV/A
 for 30A:     64< 66< 68 mV/A > constant = 5/.066 = 75.76
 for 10A:     96<100<104 mV/A > constant = 5/.1   = 50.00
 for 5A:     180<185<190 mV/A > constant = 5/.185 = 27.03
 */

const int currentPin = A7;                                  // ADC pin used for ACS712
const unsigned long sampleTime = 100000UL;    // sample over 100ms, it is an exact number of cycles for both 50Hz and 60Hz mains
const unsigned long numSamples = 250L;          // number of samples 400 microsecond each
const unsigned long sampleInterval = sampleTime/numSamples; // sampling interval, must be longer than then ADC conversion time
int adc_zero = 514;                                          // relative zero of the ACS712 for me 514 was best.  511 
int valueRead;                                                 // value read on ADC

void loop()
{
 unsigned long currentAcc = 0;
 unsigned int count = 0;
 unsigned long prevMicros = micros() - sampleInterval ;

[... 272 lines cut ...]
if [dimmer#dimvalue]<0
 TaskValueSet,12,2,0  # set lower limit for pwm value
if [9960#Gesture]=40
 TaskValueSet,12,2,[dimmer#dimvalue]+128
if [dimmer#dimvalue]>1000
 TaskValueSet,12,2,1000 # set upper limit for pwm value
if [9960#Gesture]=50
 TaskValueSet,12,2,1000 # turn on dimmer at full power
if [9960#Gesture]=60
 TaskValueSet,12,2,0    # turn off dimmer 
endon

on dimmer#pin do
Publish %sysname%/dimmer/pin,[dimmer#pin]

endon

on dimmer#dimvalue do
Publish %sysname%/dimmer/dimvalue,[dimmer#dimvalue]
EXTPWM,[dimmer#pin],[dimmer#dimvalue] # Send PIN and PWM value to promini (stored pwm values in 4 different integers at promini for each Pin via I2C if used with PME (pro mini extender) plugin
oledcmd,clear
oledcmd,on
oled,6,1,Fan#[dimmer#pin]=[dimmer#dimvalue]

endon

happytm, 2019-02-02:
I tested your sketches but did not work for me so I created following sketch & used latest plugin which work perfect for me. I think finally you have resolved this issue . I watched serial console on pro mini and it is changing integer values properly. Also I was getting proper values in to rpieasy for 3 analog values from pro mini. I very much appreciate your effort. Also can you commit following sketch for future reference before I mess it up again.

/********************************************************************************************************************\

 * Arduino project "ESP Easy" Copyright www.esp8266.nu
 *
 * This program is free software: you can redistribute it and/or modify it under the terms of the GNU 
 * General Public License as published by the Free Software Foundation, either version 3 of the License, 
 * or (at your option) any later version.
 * This program is distributed in the hope that it will be useful, but WITHOUT ANY WARRANTY; 
 * without even the implied warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
 * See the GNU General Public License for more details.
 * You received a copy of the GNU General Public License along with this program in file 'License.txt'.
 *
 * IDE download    : https://www.arduino.cc/en/Main/Software
 * ESP8266 Package : https://github.com/esp8266/Arduino
 *
 * Source Code     : https://sourceforge.net/projects/espeasy/
 * Support         : http://www.esp8266.nu
 * Discussion      : http://www.esp8266.nu/forum/
 *
 * Additional information about licensing can be found at : http://www.gnu.org/licenses
\******************************************************************************************************************/
// This file is to be loaded onto an Arduino Pro Mini so it will act as a simple IO extender to the ESP module.
// Communication between ESP and Arduino is using the I2C bus, so only two wires needed.
// It is possible to run the Pro Mini on 3V3, although the 16MHz versions do not officially support this at 16MHzl 
// By working on 3.3volt  you can skip levelconverters on I2C, but no guarantee that it will work stable.
// Arduino Mini Pro uses A4 and A5 for I2C bus. ESP I2C can be configured but they are on GPIO-4 and GPIO-5 by default.

/*
AC Light Control
 
 Updated by Robert Twomey <rtwomey@u.washington.edu>
 
 Changed zero-crossing detection to look for RISING edge rather
 than falling.  (originally it was only chopping the negative half
 of the AC wave form). 
 
 Also changed the dim_check() to turn on the Triac, leaving it on 
 until the zero_cross_detect() turn's it off.
 
 Ryan McLaughlin <ryanjmclaughlin@gmail.com>
 
 The hardware consists of an Triac to act as an A/C switch and 
 an opto-isolator to give us a zero-crossing reference.
 The software uses two interrupts to control dimming of the light.
 The first is a hardware interrupt to detect the zero-cross of
 the AC sine wave, the second is software based and always running 
 at 1/128 of the AC wave speed. After the zero-cross is detected 
 the function check to make sure the proper dimming level has been 
 reached and the light is turned on mid-wave, only providing 
 partial current and therefore dimming our AC load.
 
 Thanks to http://www.andrewkilpatrick.org/blog/?page_id=445 
 and http://www.hoelscher-hi.de/hendrik/english/dimmer.htm
 
 */
#include <Wire.h>
#include <TimerOne.h>           // Avaiable from http://www.arduino.cc/playground/Code/Timer1

#define I2C_MSG_IN_SIZE    4
#define I2C_MSG_OUT_SIZE   4

#define CMD_DIGITAL_WRITE  1
#define CMD_DIGITAL_READ   2
#define CMD_ANALOG_WRITE   3
#define CMD_ANALOG_READ    4
//#define CMD_ASK_MAX_FUNC   0x10

//volatile byte I2CReceived  = 0;
[... 272 lines cut ...]
        
        case CMD_ANALOG_READ:
        
        if (port <= 4) valueRead = analogRead(port); // port <=4 to read analog value A0,A1,A2,A3 - A4 & A5 are I2C
       // if (port > 79) valueRead = ESPrms; // port is any number >4 up to 255
        sendBuffer[0] = valueRead & 0xff;
        sendBuffer[1] = valueRead >> 8;
        
        break;
      }
  }
}

void clearSendBuffer()
{
  for(byte x=0; x < sizeof(sendBuffer); x++)
    sendBuffer[x]=0;
}

void requestEvent()
{
  Wire.write((const uint8_t*)sendBuffer,sizeof(sendBuffer));
}

Thank you.

enesbcs, 2019-02-03:
I can add it, but this sketch is using the same old I2C handling, which can lead a lockup because analogread() function located are in the receiveEvent block which is an ISR function. I am afraid that after several hour it will come about.

enesbcs, 2019-02-03:
Uploaded to https://github.com/enesbcs/ESPEasySlaves/tree/master/MiniProExtenderOACS
In case of further bugs, please open a new issue.