enesbcs/rpieasy #4: Sendtohttp command not working from rules. Works from command line.

Budman1758, 2019-01-01:
I use sendtohttp commands to start and stop video recording with Raspberry Pi cameras. Using a PIR sensor and when pir is triggered the rules sends the command.  The log shows that the pir event sends the sentohttp event but the camera does not seem to get the message.
From the tools page using the command line box it does work sending the exact same command manually.

This is the rule....

on pir#switch do
 if [pir#switch]=1  
sendtohttp 192.168.1.103,80,/cam/cmd_pipe.php?cmd=ca%201
  timerSet 1,15
endon
on Rules#Timer=1 do
sendtohttp 192.168.1.103,80,/cam/cmd_pipe.php?cmd=ca%200
endon

This is the web log output.....

22:28:34 CMD: sendtohttp 192.168.1.103,80,/cam/cmd_pipe.php?cmd==ca%201
22:28:34 CMD: timerSet 1,15
22:28:37 Event: si7021#temp=76.3
22:28:37 Event: si7021#humid=19.8
22:28:43 Event: pir#switch=0.0
22:28:49 Event: Rules#Timer=1
22:28:49 CMD: sendtohttp 192.168.1.103,80,/cam/cmd_pipe.php?cmd==ca%200
22:29:00 Event: Clock#Time=Mon,22:29

Camera should record a 15 second video. It does not.
Exact same command works from the tools command box. Camera records video.
Please let me know if you need more info.

enesbcs, 2019-01-01:
I  guess that the "=" sign processed as a condition and replaced with "==" to pass it to the python interpreter. Could you try that what the camera outputs when you write a browser to this: http://192.168.1.103:80/cam/cmd_pipe.php?cmd==ca%200

enesbcs, 2019-01-01:
I think now it is working as expected, please try it.

enesbcs, 2019-01-01:
Command box calling the same sendto function as from the rules. Please do not use two == in rules, test it in your browser directly, if it is working with the camera.

enesbcs, 2019-01-01:
1. I've made a /cam/cmd_pipe.php file on my webserver for testing, which has only one function: when called it saves its "cmd" parameter to a text file on the server with a timestamp.
2. I made the following rules
```
on test do
 sendtohttp 192.168.2.100,80,/cam/cmd_pipe.php?cmd=ca%200
endon
```
when i called "event,test" from RPIEasy than "ca 0" value arrived on the text file on the server, as expected.
At this point i have to say that i am unable to reproduce this error.

Budman1758, 2019-01-01:
Working now.  Was probably working after your update as I was missing the  comma after the port number in the url.  Sorry about that....

closed 2019-01-01 by Budman1758