enesbcs/rpieasy #8: Feature Request: Logging to external Syslog server

Budman1758, 2019-01-06:
This would be a very handy feature.  I use it for logging on all of my ESP units.  Be nice to do the same with RPI Easy.  :)

enesbcs, 2019-01-06:
I think it is possible and not so complicated. ESPEasy connects at UDP port 514 to the specified syslog server IP, but do you use Facility code and shall we expose a setting for that?

Budman1758, 2019-01-07:
> but do you use Facility code and shall we expose a setting for that?<

I don't use anything special at all.  I use Tftpd64 Service Edition on a Windows PC.

enesbcs, 2019-01-07:
I do not know what is exactly the Facility code purpose, i am using default 0. If anyone need it, it can be added later. Preliminary syslog client support added at https://github.com/enesbcs/rpieasy/commit/3f11bd58b92966ecae1de2dd374c90baab469204.

Budman1758, 2019-01-07:
Perfect!!  Just what the doctor ordered!!  Thanks.

closed 2019-01-07 by enesbcs