enesbcs/rpieasy #9: Feedback on AUTO START,MQTT & APDS9960

happytm, 2019-01-08:
I have tried rpieasy and I am very happy way it is performing.I have some feedback and questions -

- I tried BME280 and it is working perfectly.

- I also tried APDS9960 and proximity & light part of sensor work fine but I am having problem with gesture part of sensor. Do I have to choose pin for interrupt and if yes what pin should I use ? the gesture sensor  shows only 0.00 value no matter how I move my hand above it.

-  I also tried to setup paho mqtt controller and use my rpi's ip for host controller ip but log saying it is failing connection. Am I doing somthing wrong. When I install paho mqtt did I install broker or client ? if broker is not installed on pi do I have to install it separately outside of rpieasy using rpi command line using SSH ?

Thanks for your hard work.

enesbcs, 2019-01-08:
-APDS9960: If you do not provide an INT pin to the plugin it will check device status 10 times per second, and if it changes, the value changes. 
Is I2C enabled? https://m.blog.hu/bi/bitekmindenhol/image/rpi_lht0.jpg Sensor appears at I2C scan with address 0x39?
I have two of this sensors, and one of them seems buggy(?) that one can only be made to work, if i patched the official apds9960 library.. (APDS9960_LED_BOOST_300->APDS9960_LED_BOOST_100 at device.py) Can you try the proximity test python file of this library from the official site?
https://github.com/liske/python-apds9960/blob/master/rpi/test_prox.py
Itt will give you different numbers based on object distance from the sensor. (70-200mm) If it is working, than my plugin is buggy which need some fixing, otherwise your device seems buggy. (or need further patching of apds9960 lib)

-MQTT: RPIEasy only uses and installs mqtt client. You need to install mqtt broker (which serves at default port 1883), such as mosquitto onto this or other RPI (i usually install mqtt broker on the same machine as the Domoticz server), and using that IP address on the controller settings.

-The autostart feature depends on the Raspbian Linux /etc/rc.local file, which i have struggled a little, now it seems to work on all four of my test RPI devices. Did you install "screen"? Please post content of /etc/rc.local.

Thank you for your support and feedback!

happytm, 2019-01-08:
APDS9960 works fine for proximity and light So I2C settings is correct and I checked it with I2C SCAN and it shows up as 0x39 but when I set it as gesture sensor/dimmer on devices tab it is not changing value in log with my hand gestures as it should.I do not know how I change "APDS9960_LED_BOOST_300->APDS9960_LED_BOOST_100 at device.py" .
where are APDS9960 driver filed installed on my rpi ? 

MQTT now connection works after installing broker on same rpi. Thanks.

this is my /etc/rc.local file 

!/bin/sh -e
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
printf "My IP address is %s\n" "$_IP"
fi
/usr/bin/screen -d -m /home/pi/rpieasy/run.sh
exit 0

enesbcs, 2019-01-08:
Fixing LED_BOOST is a little complicated task, can you test it first with the Gesture sample program if it is also failing?
https://github.com/liske/python-apds9960/blob/master/rpi/test_gesture.py

If not working at all, you have to remove apds9960 first:
`sudo pip3 uninstall apds9960`
Then go to your home directory and clone git repo:
`git clone https://github.com/liske/python-apds9960.git`
Then try again test_gesture.py.. if not working, open device.py in the apds9960 directory with a text editor and try to replace:
`self.setLEDBoost(APDS9960_LED_BOOST_300)`
with 
`self.setLEDBoost(APDS9960_LED_BOOST_100)`
If test_gesture.py works now, then after entering the python-apds9960 you can install the modified library with:
`sudo python3 setup.py install`

Your /etc/rc.local seems OK. When your system is started, then the "My IP address is ...." line appears at the console? If yes, then rc.local is executed.
You can double check, that /home/pi/rpieasy/run.sh file is existing in this directory and make sure it is runnable, by running:
`sudo chmod a+x /home/pi/rpieasy/run.sh`
Check is screen is located in /usr/bin
`ls /usr/bin/screen`
If all the above checks, try to run directly from the console:
`/usr/bin/screen -d -m /home/pi/rpieasy/run.sh`
If there are any error messages, please let me know. Is it a Raspbian Stretch Linux?

happytm, 2019-01-08:
Fixing LED_BOOST is a little complicated task, can you test it first with the Gesture sample program if it is also failing?
https://github.com/liske/python-apds9960/blob/master/rpi/test_gesture.py

If not working at all, you have to remove apds9960 first:
sudo pip3 uninstall apds9960
Then go to your home directory and clone git repo:
git clone https://github.com/liske/python-apds9960.git
Then try again test_gesture.py.. if not working, open device.py in the apds9960 directory with a text editor and try to replace:
self.setLEDBoost(APDS9960_LED_BOOST_300)
with
self.setLEDBoost(APDS9960_LED_BOOST_100)
If test_gesture.py works now, then after entering the python-apds9960 you can install the modified library with:
sudo python3 setup.py install

**- Tried all above and it did not work so I will have to try different sensor.**

Your /etc/rc.local seems OK. When your system is started, then the "My IP address is ...." line appears at the console? If yes, then rc.local is executed.
You can double check, that /home/pi/rpieasy/run.sh file is existing in this directory and make sure it is runnable, by running:
sudo chmod a+x /home/pi/rpieasy/run.sh
Check is screen is located in /usr/bin
ls /usr/bin/screen
If all the above checks, try to run directly from the console:
/usr/bin/screen -d -m /home/pi/rpieasy/run.sh
If there are any error messages, please let me know. Is it a Raspbian Stretch Linux?

**- Ran all test above and it all test good without any error messages but it still does not start at boot up.**

enesbcs, 2019-01-08:
Yep, timer function is enabled even if not the Gesture mode wanted... its a bug, will be fixed soon.

Try the following:
`sudo systemctl status rc-local.service`
It has to write something like "Active: active (exited)", if not, please copy that here.

If no error showed, than try to start with:
```
sudo su
/etc/init.d/rc.local start
screen -r
```
The screen -r connects to the RPIEasy console if it started. Isn't it possible that some other program (apache,domoticz,lighttpd) holds port 80, and RPIEasy starts at 8080 or 8008 instead?
This will find RPIEasy in process list:
`ps -aux | grep RPI`

happytm, 2019-01-09:
Yes both port 80 and 8080 were taken by other servers  I am running and Rpieasy started automatically at port 8008.

When I used following
sudo su
/etc/init.d/rc.local start
screen -r

It started on port 80 and Rpieasy restarted on reboot. Thanks.

enesbcs, 2019-01-09:
With pleasure... yes i think this case is solved.
For the ADPS timer bug i opened a new issue, as it needs further investigation.

closed 2019-01-09 by enesbcs