enesbcs/rpieasy #10: APDS9660 timer bug

enesbcs, 2019-01-09:
In proximity/light mode the timer seems to be activated for checking gestures, and it should not work like that.

enesbcs, 2019-01-09:
I think it is now fixed.
https://github.com/enesbcs/rpieasy/commit/6e1edd66faf0a01cb91d92812cd8808471d10a09#diff-ada604af2abd7773a87bd63b99d085d4

happytm, 2019-01-10:
Is it possible to get gestures results as a whole numbers like 10,20,30,40,50 & 60 without decimal point ? I tried to see plugin code but could not figure out.

happytm, 2019-01-10:
I think even dummy sensor plugin's decimal option not working. No matter how many decimals I choose on device plugin for dummy sensor page in values section it shows in logs as number with 1 decimal only. So choosing any decimal option have no effect on how it shows in log. I am trying to get dummy sensor values as whole number without decimal.

enesbcs, 2019-01-10:
You've just found a nasty bug in the core plugin.py. In fact values were rounded to the decimals, but display and sending to controllers get a direct float no matter what.. now i am storing the converted strings instead of the floats, this way decimals should be the same everywhere. (i hope it wont trigger other errors)
https://github.com/enesbcs/rpieasy/commit/7b5900a82a255c02c69e9e94476ab18e9b81a260

You may have to delete your existing devices from RPIEasy after updating to see the differences.

happytm, 2019-01-11:
Thanks again.

I tested it and your update fixed the issue without any adverse effect so far.

enesbcs, 2019-01-11:
Thank you for your feedback, issue is closed now.

closed 2019-01-11 by enesbcs