enesbcs/rpieasy #14: MCP23017 Support

ironataerial, 2019-01-17:
RPi gpios are OK but I2C gpios are always handy when a project require multiple inputs/outputs

-D

enesbcs, 2019-01-17:
I see that it is a port extender, but i am not familiar with this device. What it can be used with az RPI? (using examples?)

ironataerial, 2019-01-17:
Each one adds 16xGPIOs with the same behavior at boot, stable and quick.
No need to worry about gpio boot state and handle all the devices with the
same logic. E.g I use 8 of the ports as output for the lawn irrigation. 4
ports as output for handling the pool filtration pump, lights and some
input pins for water level,etc.

happytm, 2019-01-17:
He is working on PME extender plugin which have plenty of extra GPIO plus ADC channels and it has it's own logic so you can load custom program on it so you have lot of flexibility on how you use GPIOs. Pro Mini extender at 3.3 V are directly compatible with RPI power supply and they are $ 2 each.

Budman1758, 2019-01-17:
It's an easy way to add 16 GPIO ports using only 2 pins on the PI as it's an I2C device.  You can potentially add up to 128 GPIO and only use 2 pins from the PI.  Very similar to the PCF8574 which has 8 pins.

Support for both of these devices would be awesome.  I have an irrigation controller device I am working on that will have support for 16 zones and I would also like to have multiple inputs so I can have lots of separate sensors for automatic control over the watering instead of only a timer only approach.

enesbcs, 2019-01-17:
RPI has 16-17 free usable GPIO's on it's own, but if you want to use a 16 zone system and other sensors, buttons at the same time, than a port extender will be indeed useful. Though i am not sure how multiple port extenders will be work at the same time. (commands especially) But i admit, it would be a nice feature.

Budman1758, 2019-01-17:
> He is working on PME extender plugin which have plenty of extra GPIO plus ADC channels and it has it's own logic so you can load custom program on it so you have lot of flexibility on how you use GPIOs. Pro Mini extender at 3.3 V are directly compatible with RPI power supply and they are $ 2 each.

That's a completely different thing than a simple port expander.  It requires another device that you have to program.  A simple port expander does not require any of that. Wire it up and off you go!

Budman1758, 2019-01-17:
> Though i am not sure how multiple port extenders will be work at the same time. (commands especially) But i admit, it would be a nice feature.

These expanders can use multiple I2C addresses.  See this link.....
https://www.letscontrolit.com/wiki/index.php?title=MCP23017

enesbcs, 2019-01-17:
I see, but how can MCPGPIO command knows, that the port is located on which expander, if multiple expanders in use?

Budman1758, 2019-01-17:
It goes by a port number range. 1-16   17-32  and so on.  How that's done in the code is way above my pay grade.  :)

enesbcs, 2019-01-18:
Thank you, than with a simple divison is can determine which extender is the destination. I just have to find my mcp at the bottom of my drawer.

Budman1758, 2019-01-26:
This may be a decent start point for the MCP23017 python code.  Might help take some of the heavy lifting off instead of starting from scratch.
https://github.com/owainm713/MCP230XX-Python-Module/blob/master/MCP230XX.py

enesbcs, 2019-01-27:
I am still fixing the PME plugin, but i've found my MCP module, and someday i can start to implement it... this code will definitely help to progress.

enesbcs, 2019-02-04:
First version added in commit https://github.com/enesbcs/rpieasy/commit/870614ea76bf293e8824097a4f4e1ab5b74eff83

Budman1758, 2019-02-05:
So far so good.  Using as output seems to be working as it should.
On, off pulse and longpulse all ok as far as I can tell.
Still need to setup a test for inputs.

In ESPEasy the pulse command is blocking. Is this true for RPIEasy?

enesbcs, 2019-02-06:
Inputs needs the Interrupt pin to be connected to the RPI for optimal usage, however input polling is also possible for >1sec intervals, but i think this is not really useful.
"Pulse" command is blocking, "mcppulse" is blocking and "extpulse" is also blocking to maintain some compatibility with the ESPEasy. :) However longpulse, mcplongpulse, extlongpulse is not blocking, they reset their values with threading system timers. (non-blocking)

enesbcs, 2019-02-18:
Please open a new issue if any problem found regarding this plugin.

closed 2019-02-18 by enesbcs