enesbcs/rpieasy #15: RPIEasy "Build" info on the main page ?

haraldtux, 2019-01-19:
To first this is a very nice project, thank you so much.

Question,

Is it possible to the RPIEasy "Build" info (RPIEasy 0.19.018) on the main page to set up ?

I had already tried it myself in to the "webserver.py" line 93 :

TXBuffer += "<TR><TD>Build:<TD>" + rpieGlobals.PROGNAME + " " + rpieGlobals.PROGVER

output RPIEasy Main page:

System Info	Value
Unit:	        84
Build:	        RPIEasy 0.19.018
Local Time:	2019-01-19 11:33:52
Uptime:	        0 days 0 hours 6 minutes
Load:	        3.41 %
Free Mem:	762864 kB

many thanks, greeting Harald

enesbcs, 2019-01-19:
Sure it can be done. Included in https://github.com/enesbcs/rpieasy/commit/201e0176079e0cfdc10568c7c6121acc4d04fee9

I realized you have a TEMPer repository, is that USB connected temperature sensor worth it's prize? I have already integrated USB-Relays through hidapi, and thinking about this TEMPer device..

haraldtux, 2019-01-19:
Thank you so much.
Yes the TEMPer's are best good sensors, only bit tricky to get them to work on linux (RPi)
But I want in briefly is see if I can make a RPiEasy plugin.
But then I me first what deepen further in the plugin structure.

Sorry for my bad English ;-)

enesbcs, 2019-01-19:
I studied the TEMPer code:
https://github.com/haraldtux/temper-q/blob/master/temper-q.py
This "for" cycle can evaluate more than one TEMPer device one-by-one?
If more than one can be used in a system at a time, than the vusb.py in the RPIEasy needs to be expanded with a new class, and the P501 plugin mixed with the P051 plugin can be a skeleton to a new TEMPER plugin. I can send you a test plugin if you provide some information about hidapi enumeration results and sample values returned by the sensor.

haraldtux, 2019-01-20:
will soon go is doing

enesbcs, 2019-01-20:
Please open a new issue, if you can provide some new information about TEMPer.

closed 2019-01-20 by enesbcs