enesbcs/rpieasy #17: Uptime

haraldtux, 2019-01-27:
is it possible to get the uptime set to 1 ?

file : commands.py
line : 453~454
   elif svname==SysVars[13]: #%uptime% 	3244 	Uptime in minutes
    return str(rpieTime.getuptime(0))
to
   elif svname==SysVars[13]: #%uptime% 	0 days 17 hours 27 minutes Uptime in day/houres/minutes
    return str(rpieTime.getuptime(1))

that reads somewhat easier, for example, on a Oled

or an additional option that you can choose, or minutes or day/houres/minutes

Thank you, 
greeting Harald

enesbcs, 2019-01-27:
After commit https://github.com/enesbcs/rpieasy/commit/e69ac666ade2dd86cb9e9833f7d568b39491b3a4
`%c_m2dhm%(%uptime%)`
can be used.

haraldtux, 2019-01-27:
Thank you :+1:

enesbcs, 2019-01-27:
No problem.

closed 2019-01-27 by enesbcs