enesbcs/rpieasy #18: GPIO command does not work ?

haraldtux, 2019-01-29:
file _P001_Switch.py    ,  line 9
#  gpio,26,1          - set pin GPIO26 to 1 (HIGH)

I've tried it on the RPiEasy page Tools/Command > gpio,4,1 and  gpio,4,0 
and in a browser http://<ip>:8080/control?cmd=gpio,4,1 / http://<ip>:8080/control?cmd=gpio,4,0

I also used fixgpio.sh

does command not work or is a bug or what am I doing wrong ?

enesbcs, 2019-01-29:
- First of all, if you want to use a GPIO pin for output, than you have to set it up for output this is the reason why Hardware->Pinout&Ports page exists. If you open that page, it's usage will be pretty straightforward. (GPIO numbering is always according to the BCM pinout number scheme.)
- Then you have to enable at least one P001 device. It will also check for GPIO dependencies, and it can not be activated until it is fulfilled
- GPIO command can be used after a P001 is enabled, and if the selected pin is an output pin
I've just tried it on my RPI Zero W, and it works as it has to be.

Please copy-paste Log from RPIEasy when reporting errors. 
In the vast majority of cases, the cause of the error is immediately visible in the log.

haraldtux, 2019-01-30:
OK, now I understand that I do not got to work !!
Thanks for the description, It works now on my RPI Zero W :+1: 

Thank you :+1:

enesbcs, 2019-01-30:
I am glad i could help.
In fact, i wrote a short description about RPIEasy usage, but it is in my mother tongue and i am too lazy to translate it to english. :)

closed 2019-01-30 by enesbcs