enesbcs/rpieasy #22: Rules timer greater than 1 do not call events

Budman1758, 2019-02-08:
I have several timers that I use in my irrigation controller.  4 actually.  Whenever a rule timer reaches the end the log shows "Rules#Timer=x"  (x being any number above 1) and nothing thereafter.  The event is not called.  Works fine if the rules timer is 1. 

Timers above 1 start and run the proper amount of time but do not call any event defined in the rules.
During the time they are running there are no other timers running concurrently.

enesbcs, 2019-02-08:
You have just found a hidden bug in the Rule engine. I think i can fixed in https://github.com/enesbcs/rpieasy/commit/470a6960d58d94c62c8623cf09c6493156c60221

Budman1758, 2019-02-08:
Looks like that bug has been squashed. :)  Good job!!

closed 2019-02-08 by enesbcs