enesbcs/rpieasy #24: New plugin: IR Receiver and Sender (38khz)

enesbcs, 2019-02-09:


happytm, 2019-02-11:
I assume you are going to use Lirc for this. I am using Lirc with Flask server for this.Some reference links below:

https://gist.github.com/prasanthj/c15a5298eb682bde34961c322c95378b#file-lirc-pi3-txt-L39

https://lb.raspberrypi.org/forums/viewtopic.php?t=173863  
https://github.com/scotty3785/pi_tv_remote ( I use this setup right now )

http://lirc.sourceforge.net/remotes/  (remote control database)
https://github.com/bbtinkerer/LircNodeJsWeb

Thanks

enesbcs, 2019-03-11:
Sadly i am unable to make it work any Python LIRC examples showed at http://www.lirc.org/api-docs/html/irw_8py-example.html Only things i got is some error messages even if i reinstalled python3-lirc package:

NameError: name 'RawConnection' is not defined

So it will be harder than i thought...

happytm, 2019-07-02:
I followed instructions at following link to install LIRC on raspberry pi 3 with stretch OS. I know some other links are not correct but link below worked for me.

https://gist.github.com/prasanthj/c15a5298eb682bde34961c322c95378b#file-lirc-pi3-txt-L39

Then I used link below to use IR transmitter and it works well. I can control all devices in the range via web.Only problem is it uses Flask server.

https://github.com/scotty3785/pi_tv_remote 

Thanks

enesbcs, 2019-07-03:
I see, but sadly the LIRC python bindings is not really useful, based on its documentation i've failed to do any simple rawread. :( The pi_tv_remote is working using the command line interface which can be OK for sending, but not usable for receiving... And the most example i've found playing around the lirc conf and a set of predefined buttons, which is not really what i need. I need rawsend and rawread function to implement something similar, that ESPEasy using.

happytm, 2019-07-27:
Is this implementation suitable for you ?

http://shallowsky.com/blog/hardware/raspberry-pi-ir-remote.html
https://github.com/akkana/scripts/tree/master/rpi

Thanks.

enesbcs, 2019-07-27:
Sorry, this LIRC examples is not suitable for me. Raw PyLIRC examples is simply not working, and generic LIRC examples works with predefined devices and KEY codes (and requires several conf files) which is not really what we need.. we need something that reads any raw IR code from any IR device... Sadly i cannot found any IR raw read example that works with Python. :(

enesbcs, 2019-10-16:
At commit https://github.com/enesbcs/rpieasy/commit/eb03a93d790d66ee480ce28e0f008143ed88e59b IR functions introduced in RPIEasy, based on these informations:
https://forum.libreelec.tv/thread/12111-kernel-module-gpio-ir-sending-ir-codes/

About the current status of IR plugins:

P016:
gpio-ir kernel module works fine, so IR Receiver plugin is mostly functional.
Decoders implemented by the Linux kernel, the RC5 significantly diverse from the values that ESPEasy show, others like NEC32,RC6,JVC looks better, but needs more testing. Please do not complain about decoder differences, as they are located in the linux kernel.

P035:
gpio-ir-tx for IR transmitting is not stable, decoded values on the other side are almost unusable, less then 10% of the output value matches for the same input, which is very sad... i am afraid 38khz signal creation by simple bitbang methods are out of the limits for a Raspberry Pi Zero.

pwm-ir-tx may be much stable for sending, but i am unable to use it at GPIO18. It needs further testing and a little help from anyone who used it before.

##Testing: (setup pins at Hardware/Pinout&Ports)
gpio-ir-tx module enabled on GPIO18:
Command: sudo ir-ctl -d /dev/lirc-tx nec32:0x8078
Result: the LED on the connected IR Transceiver is blinking, the code arrives to the other side.

pwm-ir-tx module enabled on GPIO18:
Command: sudo ir-ctl -d /dev/lirc-tx nec32:0x8078
Result: the LED on the connected IR Transceiver is not blinking!

Linux raspberrypi 4.19.75+ #1270 armv6l GNU/Linux

enesbcs, 2019-11-06:
As no feedbacks arrived it is solved now.

closed 2019-11-06 by enesbcs