enesbcs/rpieasy #27: PRiEasy don't want to run on a Raspberry Pi 3 Model A Plus

haraldtux, 2019-02-10:
Hello Alexander,

PRiEasy don't want to run on a Raspberry Pi 3 Model A Plus Rev 1.0

install :
* python3-pip 
* screen 
* alsa-utils 
* wireless-tools 
* wpasupplicant 
* zip 
* unzip
* jsonpickle
* net-tools

harald@rpi86:/ff-hier # git clone https://github.com/enesbcs/rpieasy.git
Cloning into 'rpieasy'...
remote: Enumerating objects: 367, done.
remote: Total 367 (delta 0), reused 0 (delta 0), pack-reused 367
Receiving objects: 100% (367/367), 328.26 KiB | 0 bytes/s, done.
Resolving deltas: 100% (235/235), done.

harald@rpi86:/ff-hier # cd rpieasy
harald@rpi86:/ff-hier/rpieasy # 

harald@rpi86:/ff-hier/rpieasy # sudo ./RPIEasy.py

console output : 

Traceback (most recent call last):
  File "./RPIEasy.py", line 365, in <module>
    initprogram()
  File "./RPIEasy.py", line 330, in initprogram
    hardwareInit()
  File "./RPIEasy.py", line 111, in hardwareInit
    gpios.HWPorts.readconfig()
  File "/ff-hier/rpieasy/gpios.py", line 958, in readconfig
    self.set_serial(1)
  File "/ff-hier/rpieasy/gpios.py", line 689, in set_serial
    Settings.Pinout[8]["altfunc"] = 1
IndexError: list index out of range

checked install, recently version is ok: 
* python3-dev
* python3-rpi.gpio
* python3-gpiozero
* python3-smbus
* python3-serial
* i2c-tools

harald@rpi86:/ff-hier/rpieasy # sudo ./fixgpio.sh

after all same error.

If I the (data)config files of another working RPiEasy pi (Raspberry Pi Zero W) in the data directory copy in the RPiEasy on the Raspberry Pi 3 A+ then he run 

harald@rpi86:/ff-hier/rpieasy # sudo ./RPIEasy.py 
10:42:25: Event: System#Boot
10:42:25: Webserver starting at port 8080
10:42:25: Event: Clock#Time=Sun,10:42

no further console output of erros (Console log Level: Debug Developer)

But then I can not at the http://IP:8080/hardware pages (only remains but loading)

RPiEasy runs well on a Rapberry Pi Model B Rev 2, Zero W Rev 1.1, 3 Model B Plus Rev 1.3 !!

greeting Harald

enesbcs, 2019-02-10:
Is it possible that Settings.Pinout[] is empty.  Please send me the output of the following Linux command:
`cat /proc/cpuinfo`

haraldtux, 2019-02-10:
here the output :
root@rpi86:~ # cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
Hardware        : BCM2835
Revision        : 9020e0
Serial          : 000000009679f819

enesbcs, 2019-02-10:
You are right, RPI detection is not aware of the new RPI3A+.. fixed in commit https://github.com/enesbcs/rpieasy/commit/24015b5973e28deb40124686cccc52978022ef41

haraldtux, 2019-02-10:
Thank you so much,
will be testing :+1:

haraldtux, 2019-02-10:
works well, thank you

enesbcs, 2019-02-10:
Perfect, issue solved!

closed 2019-02-10 by enesbcs