enesbcs/rpieasy #28: New plugin: TEMPer USB Thermometer

enesbcs, 2019-02-10:


haraldtux, 2019-02-11:
Maybe you have here some information to

I have these two TEMPer's
Gold TEMPer http://www.pcsensor.com/usb-thermometers/gold-temper.html
TEMPer2 http://www.pcsensor.com/usb-thermometer/temper2.html

I use these two scripts
https://github.com/marook/temper-q
https://github.com/granolamatt/TEMPer

Gold TEMPer output with this script : https://github.com/granolamatt/TEMPer
interface_number : 0
manufacturer_string : 
path : 0001:0005:00
product_id : 8455
product_string : 
release_number : 0
serial_number : 
usage : 0
usage_page : 0
vendor_id : 16701
Found the temper
path  0001:0005:01
Opening the device
Manufacturer: None
Product: None
Serial No: None
[128, 128, 9, 127, 78, 32, 0, 0]
Temp is  21.31

TEMPer2 output with this script : https://github.com/granolamatt/TEMPer
interface_number : 0
manufacturer_string : 
path : 0001:0006:00
product_id : 8455
product_string : 
release_number : 0
serial_number : 
usage : 0
usage_page : 0
vendor_id : 16701
Found the temper
path  0001:0006:01
Opening the device
Manufacturer: None
Product: None
Serial No: None
[128, 128, 7, 226, 78, 32, 0, 0]
Temp is  17.18
[128, 1, 7, 20, 78, 32, 0, 0]
Temp is  15.12

enesbcs, 2019-02-26:
If i understand well, there are several subtypes of Temper with different USB ID's... I think that this library works with more types of Temper's:
https://github.com/urwen/temper

haraldtux, 2019-02-27:
is true,  several subtypes of Temper with different USB ID's
You can try that library
I've not gotten to work that library, but will give it another try

enesbcs, 2019-02-28:
Temper support added in commit https://github.com/enesbcs/rpieasy/commit/eb765783443765ec8efe5a98984030ffe6cac420

haraldtux, 2019-02-28:
error :

Device: | Importing failed, please double check dependencies! No module named 'lib.lib_temper'

enesbcs, 2019-02-28:
ooops. i missed the library...

haraldtux, 2019-03-02:
comes very often for "Temper read error! internal temperature"
then disable the USB_Temper Device

log :
12:48:37 Event: USB_Temper#Temperature=17.6
12:49:08 Temper read error! 'internal temperature'

12:56:19 Event: USB_Temper#Temperature=19.9
12:57:17 Event: USB_Temper#Temperature=19.8
12:57:40 Temper read error! 'internal temperature'

but otherwise a nice plugin

enesbcs, 2019-03-02:
I have a TEMPerGold_V3.1 413d:1207 for testing, and i do not see this kind of glitches.
But I've removed the disabling after running an error in the loop, instead it will try to reread it in the next round.
In the meantime i found an other error: sometimes the same device gets an other usb address so i am not relying it's address anymore, just the number in the detected device list. Device settings needs to be resubmitted.

```
17:54:54: Event: System#Boot
17:54:54: Webserver starting at port 80
17:54:54: Event: Clock#Time=Sat,17:54
17:55:00: Event: Clock#Time=Sat,17:55
17:55:10: Event: temper#Temperature=26.1
17:55:25: Event: temper#Temperature=26.1
17:55:41: Event: temper#Temperature=26.1
17:55:56: Event: temper#Temperature=26.1
17:56:00: Event: Clock#Time=Sat,17:56
17:56:11: Event: temper#Temperature=26.1
17:56:27: Event: temper#Temperature=26.2
17:56:42: Event: temper#Temperature=26.2
17:56:58: Event: temper#Temperature=26.2
17:57:00: Event: Clock#Time=Sat,17:57
17:57:13: Event: temper#Temperature=26.2
17:57:28: Event: temper#Temperature=26.2
17:57:44: Event: temper#Temperature=26.2
17:57:59: Event: temper#Temperature=26.2
17:58:00: Event: Clock#Time=Sat,17:58
17:58:15: Event: temper#Temperature=26.2
17:58:30: Event: temper#Temperature=26.2
17:58:45: Event: temper#Temperature=26.2
```

haraldtux, 2019-03-02:
ok, will my TEMPer try on another RPi (with less pollution in the os), maybe it goes better on

haraldtux, 2019-03-03:
Yes, the TEMPerX_v3.3 now runs +/-12 hours with no problems :+1: 

other question : 
the TEMPerX_v3.3 has an internal and external temperature sensor.
would it be possible in the temper to make a choice option in plugin for (default)  internal temperature, and option for when you want to external temperature ?

./temper.py --json
[
    {
        "hex_data": "8080073a4e2000008001066b4e200000",
        "firmware": "TEMPerX_V3.3",
        "productid": 8455,
        "busnum": 1,
        "internal temperature": 18.5,
        "manufacturer": "",
        "product": "",
        "vendorid": 16701,
        "external temperature": 16.43,
        "devnum": 7,
        "devices": [
            "hidraw0",
            "hidraw1"
        ],
        "hex_firmware": "54454d506572585f56332e3320202020"
    }
]

enesbcs, 2019-03-03:
> the TEMPerX_v3.3 has an internal and external temperature sensor.
> would it be possible in the temper to make a choice option in plugin for (default) internal temperature, and option for when you want to external temperature ?
> "internal temperature": 18.5,
> "external temperature": 16.43,
Yes i can make it selectable in the plugin settings page.

haraldtux, 2019-03-03:
that would be nice, thanks in advance

enesbcs, 2019-03-03:
Added in commit https://github.com/enesbcs/rpieasy/commit/0e0b85500030ae2fc81da417015808b69aa650b0

haraldtux, 2019-03-04:
plugin works well, also the additional adjustment

Thank you so much Alexander :+1:

enesbcs, 2019-03-04:
no problem. issue i now closed.

closed 2019-03-04 by enesbcs