enesbcs/rpieasy #29: New feature: implement e-mail notification

enesbcs, 2019-02-10:


enesbcs, 2019-06-25:
Experimental support added in commit https://github.com/enesbcs/rpieasy/commit/730fe678232dcbc2e5c7d5056dd0637ac2331126

Budman1758, 2019-06-27:
Can't seem to get this working.  Pretty sure the settings page for the email are correct. But it seems like no matter how I try to get the program to send an email nothing happens.
No matter what log level is set to below is all that shows when attempting to "fire" the email. 
Tried "event,notify1"  from command line in the tools page.
Also same command using a rule.  Same result.  Ideas?

Jun 26 18:03:19	192.168.1.10	RPIEasy	kern	notice	Timekeeper	CMD: event,notify 1
Jun 26 18:03:19	192.168.1.10	RPIEasy	kern	notice	Timekeeper	Event: notify 1
Jun 26 18:03:21	192.168.1.10	RPIEasy	kern	notice	Timekeeper	CMD: event,notify 1
Jun 26 18:03:21	192.168.1.10	RPIEasy	kern	notice	Timekeeper	Event: notify 1

enesbcs, 2019-06-27:
Did you try simple "notify 1" command without event?
https://www.letscontrolit.com/wiki/index.php/EasyNotifications

Budman1758, 2019-06-27:
After further review of my email server settings I got it to work.  Also correct that the "event" command not needed.  Will be doing more testing but so far, so good.  :)

enesbcs, 2019-07-15:
Thank you for your feedback, issue is now closed. Feel free to reopen in case of any related bug appears.

closed 2019-07-15 by enesbcs