enesbcs/rpieasy #30: Node-Red front end

happytm, 2019-02-10:
You are doing all new plugins at amazing speed. I don't know how you can do this all this fast. I  am working on front end part for RPIeasy to control devices using node-red. I have basic front end working so user of rpieasy can install node-red and easily connect to rpieasy and have any rpieasy  or espeasy device auto detected and basic GUI created by just adding ip address of device on node-red dashboard.  Now user can start monitoring sensor data and control whatever is connected to device without any further knowledge of node-red or they can further I would like to ask you if it is possible to add node-red on dependency  page as add on dependency preferably in rpieasy folder to make it smooth process for user to install it and then it can be connected to rpieasy via controller page using MQTT as a new controller.That way RPIEasy is complete package working as out of box IOT platform. With node-red's powerful logic engine and 300+ plugins library (they call it nodes & flows)  it will be very easy to create whole house IOT automation without knowing programming. I need your feedback and I will request you to set up node-red so you can test it. This is not urgent and I do not want to sidetrack you from your core mission. Just for reference  I left following comment at nod-red. 

https://discourse.nodered.org/t/new-home-dashboard-using-uibuilder-and-bootstrap-vue/6764/6

Thanks
Ken

enesbcs, 2019-02-11:
Someday i have more time than other.. and some plugins can be easily added based on another one. :)

Sorry, I do not use NodeRed, as i do no need it. I can do anything in Domoticz with the in-built LUA engine, if not, i can even write an own Python Plugin for Domoticz, like i did it for my Shelly1.
NPM and Javascript is the two most CPU intensive stuff i know. However i can setup a test installation in an Oracle Virtualbox someday.
I am not sure what you mean with controller page and NodeRed, as i know, the Generic MQTT should work with NodeRed. NodeRed may be added as an optional dependency, but i have to learn more of it's installation method before. 
%sysname%, %systime% and other Global Variables defined in commands.py and used by several places in the program. I implemented followed this list (as i can), when coded:
https://www.letscontrolit.com/wiki/index.php/ESPEasy_System_Variables

However in the Generic MQTT controller, currently only two template can be used: %tskname% and %valname%, this controller can send and receive data if the standard %tskname%/%valname% template is used in topic, in other cases, only 1 way is possible: sending.
If you list what system variables you want to use, i will try to implement them.

happytm, 2019-02-11:
I have tried Home assistant and Openhab but they looked overkill for my simple setup.I am going to try domoticz now to see if it is suitable for me. My other request for adding system variable was related to further work on node-red but right now let me try domoticz first. I want to keep my setup  as simple as possible. I am interested in saving data as I am going to do  solar project and want to analyse solar data for my area before I spend lot of money on it.  

Thanks

enesbcs, 2019-02-11:
As i know Domoticz has a 5 minute period for saving sensor data. However i am satisfied with basic diagrams that Domoticz provides me, i admit that in some cases it is inadequate for someone who needs data at more frequent intervals or longer time period. I like Domoticz especially because it uses 3-7% of my  RPI Zero CPU (1 core 1Ghz) with 180MB RAM.
For advanced logging NodeRed+SQL i guess that a more powerful machine may be needed. :)

closed 2019-02-12 by happytm