enesbcs/rpieasy #31: Rules timer bug.

Budman1758, 2019-02-11:
I think I have found another minor bug in the rules engine.  When you have a timer running and want to cancel it for whatever reason the command is is to set the timer to 0.  (timerset,1,0)
What happens is the command is recognized but the result in the log is Event: Rules#timer=1.  This does not cancel the rule.  It has the effect of just shortening it because what ever is supposed to happen when the timer expires still does.  

In short, the timer is not actually cancelled. Whatever logic is defined to occur when it expires naturally is still executed.

enesbcs, 2019-02-11:
yes, as i see timerset,1,0 does not cancels the timer, instead trigger it's event.

enesbcs, 2019-02-11:
I think i fixed it in commit https://github.com/enesbcs/rpieasy/commit/cd9eadc5a5a19134dd4acccef784940de11e26d8 Please try it.

Budman1758, 2019-02-12:
You must be an Exterminator in your day job.  You're pretty fast at killing those bugs.  :>)

enesbcs, 2019-02-12:
In cases when i have some ideas, yes!  :)

closed 2019-02-12 by enesbcs