enesbcs/rpieasy #32: Clock#Time rules bug. Only 1 Clock#Time event executes.

Budman1758, 2019-02-11:
I seem to have either found a bug or a limitation. If you have more than one Clock#Time event in your ruleset only the first listed one of them will execute.  Was tearing my hair out (of which I have VERY little left) until I tried commenting the first one of them out. When I did that then the second one in the rules worked fine.  Even if the first one will do nothing logic wise the second one is still ignored.

Is this a bug or a limitation?

enesbcs, 2019-02-11:
i assume this is a bug. I've only tested with one timer in the rush, i have to look after the rule engine...

enesbcs, 2019-02-11:
I think i fixed it in commit https://github.com/enesbcs/rpieasy/commit/cd9eadc5a5a19134dd4acccef784940de11e26d8 Please try it.

Budman1758, 2019-02-12:
Looks like that bug is squashed.  Good Job!!

enesbcs, 2019-02-12:
Thanx, issue solved!

closed 2019-02-12 by enesbcs