enesbcs/rpieasy #33: " and " in rules either not implemented or not working.

Budman1758, 2019-02-12:
I have a rule I am trying to use with an "and" statement in it. I know this was implemented in ESPEasy but not sure about here.

The rule I am using....
`if [station#hours]>47 and [auto#switch]=0 do`  This evaluates as false. The switch is indeed at "0" and the station hours are indeed more than 47
`if [station#hours]>47`   Evals as true by itself
`if [auto#switch]=0`   Evals as true by itself.
 Like the issue title says. Either not implemented or not working.
If not implemented can you?  :)
If a bug, can you exterminate? :)
No big rush. Whenever you have the time.....
Thanks again for this awesome software!!

enesbcs, 2019-02-12:
It's interesting as "and" worked in my first testings.. could you switch Log level to Debug and see "Parsed condition" lines? I will check it today evening. /it's morning now here :) /

Budman1758, 2019-02-12:
"Station" is a dummy device where I store values. The hours value is incremented up 1 every hour. After a set time this value is used to start the watering sequence.  So [station#hours] is just a dummy value.
"Auto" is a switch connected to a PI GPIO. "auto" is the switch name and "switch" is the value name. So [auto#switch is simply a switch that is either 1 or 0. (input pullup) in the hardware settings.

`192.168.1.62	Feb 11 22:35:00	RPIEasy	kern	debug	Irrigation	Parsed condition: if 48>47 and 0=0 False`

Debug Developer Level.  Above line is all that shows.  I get that via syslog. I do not have serial connected. If need be I can set that up.

enesbcs, 2019-02-12:
== is not equal to = in python.. using eval() is convenient choice, but equal signes needs to be converted back and forth.. i think it is fixed in commit https://github.com/enesbcs/rpieasy/commit/60ec0f8cecb8f8a098f7d6f0a593403158f611e7

Budman1758, 2019-02-12:
Another bug seems to have been exterminated!!

enesbcs, 2019-02-13:
I am afraid there are some more hidden bugs.. But thanks, this one closed.

closed 2019-02-13 by enesbcs