enesbcs/rpieasy #34: RPI GPIO input changes do not always register. No event.

Budman1758, 2019-02-13:
I have been noticing this quite a bit during the conversion of my irrigation controller from ESP to RPIEasy.  I have 3 toggle switches connected to 3 of the PI GPIO pins.  Quite a bit of rules logic happens when these switches are thrown.  It seems random but its more noticeable when the unit has been sitting idle for a period of time.  When the switch is toggled the PI does not react at all. Nothing in the log output at all when this happens.  However toggling it within a few seconds of the firs attempt it will work as it should.  It seems like the PI doesn't "get the message" sometimes.

Te actual GPIO's in question here are 17, 22 and 27.  All are set as "input pullup" in the hardware page.
Could it be that the missed pin change is due to the PI doing something else at that instant? Like I mention above its random but noticeably more after some idle time.

enesbcs, 2019-02-14:
I met this error while testing "and" rule line evaluation, found that the 200msec bouncetime that i've used in P001_Switch plugin may be too much sometimes so i decreased it to 10 msec in the last commit.
If it also produces this error, i can decrease it to 0, but in this mode, the pin values can change more than once for one buttonpress.

Budman1758, 2019-02-14:
OK.  Just updated and will see if that helps.

enesbcs, 2019-02-14:
I've removed bounce time completely in commit https://github.com/enesbcs/rpieasy/commit/3880fc683806f6c107f5479c9ff4b4058d6141c5 from P001. In theory RPi.GPIO library callbacks calls p001_handler() function in my plugin instantly if changes occured. If for some reason a callback can not be registered than the log file contains an error message and tries to check GPIO state ten time per second for fallback method. (i guess i can made this fallback method selectable in the plugin page for testing)

Budman1758, 2019-02-15:
OK.  Will test and report back here.

Budman1758, 2019-02-15:
Well.  So far so good. I haven't seen the issue now for some time. Guess we can close this now.

enesbcs, 2019-02-15:
OK. If anti-bouncing will be needed in the future for any reason, i can add an option in the plugin settings.

closed 2019-02-15 by enesbcs