enesbcs/rpieasy #35: After the update 0.19.43 deshboard.esp problems

haraldtux, 2019-02-14:
After the update 0.19.43 deshboard.esp problems

with this version 0.19.41, 
with buttons with this url link
http://IP:8080/dashboard.esp?cmd=OLEDFRAMEDCMD,on
http://IP:8080/dashboard.esp?cmd=MCPGPIO,1,1

After the update 0.19.43
no buttons, but this url link
http://IP:8080/="dashboard.esp?cmd==OLEDFRAMEDCMD,on"
http://IP:8080/="dashboard.esp?cmd==MCPGPIO,1,1"

and further colors other than the specified html colors
#c80a41 = red , but now green

enesbcs, 2019-02-14:
Please try commit https://github.com/enesbcs/rpieasy/commit/3880fc683806f6c107f5479c9ff4b4058d6141c5. This should solve the equality sign issue... but the color thing is a little fuzzy to me.

haraldtux, 2019-02-15:
It works again, the colors and buttons are also 👍

enesbcs, 2019-02-15:
Perfect, issue closed.

closed 2019-02-15 by enesbcs