enesbcs/rpieasy #40: New plugin: MH-z19 Co2

enesbcs, 2019-02-21:
asked in https://github.com/enesbcs/rpieasy/issues/37
Currently i do not have this device, and is above my order threshold.

enesbcs, 2019-02-23:
I have ordered such a device using the donations i've received for RPIEasy project.
Thanks to happytm and Budman1758!
I've found a python library that let me implement this plugin easily - i hope:
https://pypi.org/project/mh-z19/

enesbcs, 2019-03-13:
Device arrived to me, the pypi library seems to work but after some time the co2 results always stucks at 5000... i've disabled ABC logic, but nothing helps, unless i power-cycle the mhz-19 when it starts to report 410ppm co2.. for a while. 
I've order this: (which looks like an A version)
https://www.banggood.com/MH-Z19-0-5000PPM-Infrared-CO2-Sensor-For-CO2-Indoor-Air-Quality-Monitor-UARTPWM-p-1094463.html?rmmds=myorder&cur_warehouse=CN
But I've got a B version, which misses 2 pin. Looks like unusable in this form.. anybody know, if it can be used stable?

enesbcs, 2019-03-14:
I had to execute zero_point_calibration() once and it's seems to solve this error. Plugin creation is in progress.

TD-er, 2019-03-14:
Do not use zero point calibration!!!
You will very likely ruin the sensor.

The main difference between the A and B version is that he A version outputs some kind of uncertainty level indicator. 
If it is '64', then all is OK.
The "B" version does output '0' on that position.

The max output value (5000 in your case so it seems) is sent when the sensor is booting and initializing.
You may better leave the sensor alone during this process. Do not send any command to it to execute as long as it is in that state.

enesbcs, 2019-03-14:
Too late, i am already done it. It seems very hopeless to me, that previous night the sensor worked after powering up returning by 410-420ppm 26C, but only for 30 seconds! After that, only 5000ppm's arrived from it, even if i let it for an hour! I doubt that initialization can take this ling. I've disabled ABC on several occasions but nothing helped, until i executed the calibration. After the calibration finished, the results goes back to 400-410ppm and if i breath to it, values raises up to 500ppm, than if i leave it alone, it slowly goes back to 400ppm so i guess it is working. And it's output is now stable at least.

TD-er, 2019-03-14:
Things to check with this sensor:
- No IR source near it (also no sunlight)
- Make sure it has enough power
- Run it at 5V (GPIO pin output level is 3.3V)
- Put it in a room (at normal room temperature) with ABC enabled (= default setting) and leave it on for 24h.

And remember, do not (!!!!!!) use the zero point calibration.
You will very likely ruin the sensor.

Only use it when the sensor has been in a well ventilated room, no direct sunlight and with some plants in it, during the day.
It also has to be like that for at least 20 minutes.

enesbcs, 2019-03-14:
Thank you, good advices. It has 5V2A which has to be enough, it is night here and no sunlight. /honestly most of my tests occurs after work hours at home, so every time i testing these stuffs without sunlight :) / I guess i can recalibrate it sometimes with some plants and opened windows at summer time, now it's stable output is enough for testing purposes to me.

TD-er, 2019-03-14:
Just put it in a room where no people are for a while (also no animals ;) ) and during the day with plants present (not needed, but will help get he CO2 levels down)
The reason why you want to do that during the day, is because the plants will collect CO2 when the sun shines and output CO2 when the sun is set.
You just have to make sure the sun is not directly visible by the CO2 sensor, since it operated via IR absorption. The air with CO2 in it will absorb more IR light than the path of light through some air with less CO2.
The difference is some indicator of CO2 concentration.

It is really hard to get the CO2 level down to 400 ppm when there are CO2 producers in the room :)

And it is not only the power supply which may cause issues. Also the USB cable has to be one that can handle some current without causing a voltage drop.

enesbcs, 2019-03-14:
MH-Z19 plugin added to RPIEasy at commit https://github.com/enesbcs/rpieasy/commit/34807b9c46e6607925bda74bf4fb7807dda1b9f2 for testing purposes.
It's using RPI's hardware serial, it is a little tricky to enable with 'raspi-config' and MH-Z19 sometimes does not reply, but in overall, it's just work.

enesbcs, 2019-03-30:
Implementation issue is closed now.

closed 2019-03-30 by enesbcs