enesbcs/rpieasy #43: rpieasy-cli-tools

haraldtux, 2019-02-23:
Hello all , 
for the enthusiast

rpieasy-cli-tools command line menu script for RPiEasy written in Python

https://github.com/haraldtux/rpieasy-cli-tools

have fun, greeting Harald

enesbcs, 2019-02-24:
Thank you! Is it replacing your previous rpieasy-update.sh script? I will add it to the readme and the wiki. As i see your method will work in case autostart is not enabled in the RPIEasy, as it is using run.sh with an endless loop which is not easy to stop. :)

haraldtux, 2019-02-24:
Yes, it has been replaced by my previous rpieasy-update.sh script.

I will also see if I can find a solution for run.sh

to be honest I had not realized, I use that run.sh script itself not

enesbcs, 2019-02-24:
In the integrated update method at Tools page i just downloading the files and overwriting them, while they are running, then do a stop (exit). The run.sh does the restarting on its own. Readme updated.

closed 2019-02-24 by enesbcs