enesbcs/rpieasy #44: New plugin: VL53L0X distance meter

enesbcs, 2019-02-28:
Library for implementation: (preliminary testing needed)
https://pypi.org/project/VL53L0X/

enesbcs, 2019-03-11:
Back to square 1, pip3 install VL53L0X output: "No matching distribution found for VL53L0X"

Budman1758, 2019-03-11:
I just downloaded the files from the link in your first post.  Maybe they are just not in the python package index for some reason.

enesbcs, 2019-03-11:
Maybe, but if it's not in the index than it would not be so easy to write a dependency check and autoinstaller for it. In the meantime i've found an other python sample, which may be more useful, but needs some rewriting.
https://github.com/DexterInd/DI_Sensors/blob/master/Python/di_sensors/VL53L0X.py

enesbcs, 2019-03-12:
Test version added in commit https://github.com/enesbcs/rpieasy/commit/44392baeb54be6da2822bcc8b3d8f3a06ee1ab9e

enesbcs, 2019-03-30:
Implementation issue is closed now.

closed 2019-03-30 by enesbcs