enesbcs/rpieasy #49: New plugin: SR04 range finder

enesbcs, 2019-03-15:


TD-er, 2019-03-15:
Please note that some of these need some extra electronics like a level converter.
Or else you may damage GPIO pins.
And for the Arduino environment we use the "newping" library, which appeared to be way more stable compared to the previous implementation.

enesbcs, 2019-03-15:
I know, already used an SR04 with ESPEasy for presence detection in restroom, with a voltage divider. (2 resistor)
For Linux/Python the newping library is not an option, and i think a multitasking Linux OS will not provide as accurate timings as an ESP8266/Arduino hardware. Despite that my first test results seems fine in pure python. Now i will test it with multithreading code.

enesbcs, 2019-03-15:
Support added in commit https://github.com/enesbcs/rpieasy/commit/3d697ce638291ab25d29b3e42e70f0875f95fa77. Seems to works ok in multithreaded environment. I have experienced 1-4mm differences between measures with physically same distances. There were some big values in multiple measures, but with 5-8 samples at once the plugin can filter them - mostly.

enesbcs, 2019-03-30:
Implementation issue is closed now.

closed 2019-03-30 by enesbcs