enesbcs/rpieasy #55: New plugin: LM75A

haraldtux, 2019-03-28:
Hello Alexander 
Can you make a LM75 I2C temperature sensor plugin

https://nl.aliexpress.com/item/LM75A-Temperature-Sensor-I2C-Interface-Temperature-Module/32685410585.html

Working Python3 code : 

#!/usr/bin/env python3
#
## Raspberry Pi LM75A I2C temperature sample code.
# 

import sys
import smbus

# By default the address of LM75A is set to 0x48
# aka A0, A1, and A2 are set to GND (0v).
address = 0x48

# Check if another address has been specified
if 1 < len(sys.argv):
	address = int(sys.argv[1], 16)

# Read I2C data and calculate temperature
bus = smbus.SMBus(1)
raw = bus.read_word_data(address, 0) & 0xFFFF
raw = ((raw << 8) & 0xFF00) + (raw >> 8)
temperature = (raw / 32.0) / 8.0+ -1.3
# Print temperature
print ('LM75 Temperature: {0:0.2f} *C'.format(temperature))

enesbcs, 2019-03-29:
I can surely do it, but this device has a +-2Celsius precision which is very low. I can not imagine why anyone choose this type instead of HTU21D/Si7021/DHT22/AM2320/BME280...

haraldtux, 2019-03-29:
Indeed, there you have also equal...
I was thinking that the LM75 also in a USB TEMPer Gold and X are and they are cheap.

is also but one question, If you think it has no added value, it is good, then we close this issues :+1:

enesbcs, 2019-03-29:
as you can add the necessarry Python code i can do it with minimal work :)

haraldtux, 2019-03-29:
yes, I have a few of the LM75 device (see url) so I would the plugin be able to test

enesbcs, 2019-03-29:
Hi Harald!
Your code is now embedded in commit https://github.com/enesbcs/rpieasy/commit/d0f2b637346773b099e7016c1621f6f47118405a. Please test it. :)

haraldtux, 2019-03-30:
the LM75 plugin work very well ! :+1: 
log : 
09:18:03 Event: LM75#Temperature=20.0
09:18:04 Event: LM75#Temperature=19.9
09:18:05 Event: System_info#Uptime=10374.0
09:18:05 Event: LM75#Temperature=20.0
09:18:05 Event: System_info#Wifi_RSSI=-47.0
09:18:05 Event: System_info#System_load=5.6
09:18:05 Event: System_info#CPU_Temp=39.7
09:18:06 Event: LM75#Temperature=19.9
09:18:07 Event: DHT22#Temperature=19.8
09:18:07 Event: DHT22#Humidity=41.8
09:18:08 Event: LM75#Temperature=20.0
09:18:09 Event: LM75#Temperature=20.0

Thank you Alexander

haraldtux, 2019-03-30:
can you in _P069_LM75A. py line 61 set to "8" , I have a LM75 sensor at address 4f stand, which I can now select not.
If I change I do have the list of addresses

webserver.addFormSelector("Address","plugin_069_addr",8,options,optionvalues,None,int(choice1))

haraldtux, 2019-03-30:
would you be able to fit the file _P069_LM75A.py in line 61 from 2 to 8 , so that you can choose from 8 addresses (have a LM75 module with the 4F address
from : 
webserver.addFormSelector("Address","plugin_069_addr",2,options,optionvalues,None,int(choice1))
to : 
webserver.addFormSelector("Address","plugin_069_addr",8,options,optionvalues,None,int(choice1))

has no hurry, may best for next update ;-)

enesbcs, 2019-03-30:
Ehm.. source is copied from the AM2320 plugin. :) Fixed in commit https://github.com/enesbcs/rpieasy/commit/6d8961bf6591dbd879ea3d6ac803196dae139896

haraldtux, 2019-03-30:
Thank you :+1: 

can be closed !

closed 2019-03-30 by enesbcs