enesbcs/rpieasy #56: feature request : LANC Protocol for camera control

happytm, 2019-03-29:
Is it possible to add plugin for camera control using LANC Protocol described in following link?  The arduino code is meant for esp8266 but I do not now how much challenging it would be for you to port it to  rpieasy? I was planning to use it to control  only On/Off and Zoom In/Zoom Out functions on camcorder  remotely from web interface.

enesbcs, 2019-03-30:
It may be possible, but needs some investigation. Based on this: http://www.boehmel.de/lanc.htm it is like RS232 9600 baud communication, but also mentions the presence of 5-8VDC on the line, so a hardware level converter will be a musthave for connecting an RPI (3.3V I/O) to a camera. Did you use the Blackmagic-LANC sketch previously?

happytm, 2019-03-31:
Yes I was going to use level converter. I have not tested Blackmagic sketch as I am waiting for interface cable. 

Thanks

enesbcs, 2019-03-31:
Python's time.sleep() functions best precision is normally in millisec range not in usec.. but then LANC communication demands high precicion timing. I've made a sample test python application that you can try.
https://github.com/enesbcs/lanc-python
It's basicly working in the same way as the Blackmagic-LANC arduino sketch. GPIO17 is the receiver pin and GPIO27 is the sender pin, but can be changed.

happytm, 2019-03-31:
Thanks.

I will test as soon as I get cable and willpost feedback here.

enesbcs, 2019-10-02:
Case is now closed as no feedback provided, feel free to reopen if needed.

closed 2019-10-02 by enesbcs

JoseFanc, 2023-03-10:
Hi.  thanks for your script  @enesbcs  . Actually i need control a BMCC Camera  with Raspberry pi3 in order to start and stop  REC  into interval time.. Did you use a Raspberry pi to control this Camera? Because i think that the GPIO work up to 3v3 and 16mA as input or output.  And in internet i found information about LANC protocol and in all cases used to work between 5-8v

enesbcs, 2023-03-11:
I do not have such camera, so i provided this script for anybody who has one for testing.

ZhouYixun, 2024-05-07:
@enesbcs SONY has a 2.5mm audio port for the lanc protocol, and a microusb port for the mutli protocol, which both can be used to give commands such as taking pictures to external controllers. Now I have an external controller, but I don't know how the Raspberry PI gets that signal, because my camera make by Pi, Do you have any ideas？