enesbcs/rpieasy #58: Feature Request - 28BYJ-48 Stepper Motor

happytm, 2019-04-12:
I have been working on Pan/Tilt platform for my Sony camcorder using PI and following inexpensive motors.There will be 2 motors  for pan and tilt. 

https://www.ebay.com/itm/ULN2003-28BYJ-48-5V-Step-Motor-Gear-Stepper-Motor-4-Phase-Step-Motor-for-arduino/372633711963?hash=item56c2b3355b:m:m54JzxcGoAzTjBQwmzYBy6Q

I found couple of people on youtube who are using these motors with python code.

https://www.youtube.com/watch?v=Dc16mKFA7Fo

https://www.youtube.com/watch?v=zuiYdYUgyXk
Python Code : https://github.com/Basch3000/raspberry-pi-projects

Is it possible to make plugin for rthese motors ?

Thanks
happytm

enesbcs, 2019-04-13:
Hi,
Yes i think it is absolutely possible, i just have ordered it for testing.
The python code seems good, do you have any ideas about the necessary (Rules) command semantics? Such as "motor,<motorname>,<direction>,<degree>"

happytm, 2019-04-14:
I think it would be better if we use command structure like below:

- For pan control command could be ptz,pan,left or right,and optionally degree in number for fine 
  control or heading like HOME,LAST POSITION,N,S,E,W,NE,SE,NW,SW etc.
- For tilt control command could be like ptz,up or down and optionally degree number from 0 to about 
  150

enesbcs, 2019-05-16:
I'm currently playing with this stepper motor, but as i read at many places (and experienced that) "Steppers don't have a home position". So this will be a little tricky, if not impossible.
I can try to maintain the relative position, but the home position might be the one that seems 0 at the program startup.
As i see, i can make commands like i wrote before: set position to a specific angle/direction (can be made some alias like N, S, W, E...) and speed of motion. Continous slow loop between two position is also seems an achievable goal.

enesbcs, 2019-05-20:
Support added in commit https://github.com/enesbcs/rpieasy/commit/f96254db0554a803df1949f2400d39882e4bcee4

Commands:
motor,"taskname",pos,"angle"    - Set the motor named "taskname" to "angle" (0-360) position from a fixed ZERO position, angle can be number between 0-360 or HOME,LAST,N,E,S,W

motor,"taskname",setzero        - Set the current position of the motor as ZERO

motor,"taskname",left,"angle"   - Rotate the motor named "taskname" with "angle" (0-360) from current position - Anti-clockwise

motor,"taskname",right,"angle"  - Rotate the motor named "taskname" with "angle" (0-360) from current position - Clockwise

motor,"taskname",pan,"startangle","stopangle","delay","speed"
  - Pan the motor named "taskname" continously between "startangle" and "stopangle", staying off for "delay" second in between and speed can be 1-20

motor,"taskname",off            - Sets the motor off - pull all 4 gpio to low (otherwise it consumes energy while standing)also stops panning immediately but relative position can be messed if used, use panstop instead

motor,"taskname",panstop         - Sets the motor stop panning after finishing it's current move

enesbcs, 2019-07-17:
Issue is closed, please feel free to reopen in case of related bugs.

happytm, 2019-07-26:
I tested all the command available and it works fine. For pan motor movements the position of motor shaft has to be pointing down for right clockwise and anticlockwise moves to work properly.

It will be useful to add following 2 commands to conserve power if it is possible:

motor,all,off
motor,all,stop

Thanks.

enesbcs, 2019-07-27:
The simple off and panstop commands is not enough? You mean "motor,all" as a command that scans through every enabled Stepper plugin and executes the off or panstop command each of them?
Otherwise If needed i can modify the stepper motor to accept floats... :)

happytm, 2019-07-27:
When I think about it for solar tracker or camera pan/tilt there are only 2 motors so in rules all I have to do is issue off command for each motor separately so that will be enough.

Thanks

enesbcs, 2019-07-29:
After today commit https://github.com/enesbcs/rpieasy/commit/154d6f3bb9dd7848ad8c1b96b1626a7a32575062 the "angle" parameter can be float and motor,all,off and motor,all,panstop commands works on all enabled stepper motors at the same time.

happytm, 2019-07-30:
Thank you for float value.Also both new commands works as expected.It will save some lines in rules code.

Thanks again

closed 2019-07-30 by enesbcs