enesbcs/rpieasy #59: Main pages icons

haraldtux, 2019-04-22:
Hello Alexander and RPiEasy users,
was been at the arts and crafts, maybe you find it nice icons in the main pages

[image]

file : rpieGlobals.py
code (line 14~22): 
`gpMenu = []`
`gpMenu.append(['\u2302 Main','.'])`
`gpMenu.append(['\u2699 Config','config'])`
`gpMenu.append(['\U0001f4ac Controllers','controllers'])`
`gpMenu.append(['\U0001f4cc Hardware','hardware'])`
`gpMenu.append(['\U0001f50c Devices','devices'])`
`gpMenu.append(['\u29f4 Rules','rules'])`
`gpMenu.append(['\u2709 Notifications','notifications'])`
`gpMenu.append(['\U0001f527 Tools','tools'])`

Icon list : https://www.charbase.com/block/miscellaneous-symbols-and-pictographs

btw:
with Linux Debian Chromium Web Browser not all icons are displayed (emember that I have a somewhat older Debian ver. turn)
with Windows 10 Chrome Web Browse It is good

TD-er, 2019-04-22:
These are the icons used currently in ESPeasy, and those were chosen because we simply needed some icons for the view of tabs in the small view of mobiles.
By using those icons, you simply don't need to have images embedded in the firmware, which saves space.

I guess that if you find better ones to be used in RpiEasy, we should also use them in ESPeasy.

enesbcs, 2019-04-22:
In fact in RPIEasy any png images can be added to the html as there are a lot of memory and SD card place available.. but i really like this minimalistic approach. :) Thanks for contributing, added with todays commit to RPIEasy!

enesbcs, 2019-04-22:
Generally i think that the current icons is fine. Altough i can imagine these
- for Rules https://www.charbase.com/1d11a-unicode-musical-symbol-five-line-staff
- for Hardware: https://www.charbase.com/2328-unicode-keyboard
Maybe it would be a good idea to put a vote/survey on this in the letscontrolit forum?

haraldtux, 2019-04-22:
nice, It works !!

Thank you

enesbcs, 2019-05-07:
Issue closed as no other feedbacks provided. Feel free to reopen in case of related ideas.

closed 2019-05-07 by enesbcs