enesbcs/rpieasy #65: Feature Request: Pulse counter plugin

enesbcs, 2019-05-25:


enesbcs, 2019-05-26:
Plugin added in commit https://github.com/enesbcs/rpieasy/commit/c546140b1bf075002732b5fe78ef4bed84308e46

bennybubble, 2019-06-09:
Hi,

Finally I had time to test the plugin and it seems to work fine.
Applied a pulse of 10 Hz and the RPI (model A - V1.1) was able to count all pulses.

Thanks !

-ben

enesbcs, 2019-06-10:
Thank you for your feedback, please reopen this issue, in case of pulse plugin bugs appears.

bennybubble, 2019-11-12:
Hi enesbcs,

I'm using the pulse counter for a while now but think there is  a problem with it:

I'm counting pulses of my gasmeter on rising (or falling) edge. 
A TCR5000 sensor is mounted in front of the meter and gives 1 pulse per cubical dm of gas (there is a reflective part on the right digit of the meter).  The pulse has been verified with a scope and there is NO bouncing or any disturbance.
Due to the gas usage, the meter can only give 5 pulses (max) per minute. Pulse duration is ~1.5 seconds.
Problem: Sometimes I see readings of 60 pulses (!) per minute, most likely this happened when the meter stops and the output stays HIGH (instead of LOW-HIGH-LOW).
Debounce time was set to 1000 mS to ensure there was no bouncing and it looks like there is a link between the measurement period (1 minute), the number of pulses (60) and debounce time(1 sec).

[image]

I looked at the pulse code but could not figure out how it works perhaps you can check this again!

-ben

enesbcs, 2019-11-13:
Interesting. Perhaps i can add debounce time value to the Rpi.GPIO add_event_detect instead.

bennybubble, 2019-11-13:
It looks like it does detect level instead of rising/falling edge.

enesbcs, 2019-11-13:
For counter type did you use "BOTH","RISING" or "FALLING"?

bennybubble, 2019-11-13:
I tried both FALLING and RISING since BOTH would give me a number which is twice as high.

enesbcs, 2019-11-14:
I've added extra checks for the event handler in commit https://github.com/enesbcs/rpieasy/commit/1d3f82cfce325383969c174e6b468209518537b0 . I am not sure if it helps, but i have no other idea.

enesbcs, 2019-11-21:
Closed as no further feedbacks received.

closed 2019-11-21 by enesbcs