enesbcs/rpieasy #67: Feature Request: FHEM as a controller

pirquadrat, 2019-05-28:
Hi,

according to the topic on Let's Control It forum, it would be great if you could provide FHEM support. Some API things can be found here: https://fhem.de/commandref.html (I think the first four points _introduction_, _FHEM command types_, _device specification_ and _attributes_ will be relevant. whole documentation is veeery long).

Thank you in advance!

enesbcs, 2019-05-28:
Sorry, the web page you've appended has no relevant information to me. I can understand the Domoticz API from this: https://www.domoticz.com/wiki/Domoticz_API/JSON_URL%27s but frankly i have no clue what to do with this: https://fhem.de/commandref.html Call this commands from JSON? On what port and URL? Example usage?

pirquadrat, 2019-05-28:
Hey, now I got what do you mean. An example for a HTTP command is: https://fhem:8083/fhem?cmd=set%20ESPEasy_1_switch_1%20on&fwcsrf=csrf_1166923445179493, for turning on the switch_1 on ESPEasy_1. That digits after csrf is a token what is being generated with every start of fhem, more here: https://wiki.fhem.de/wiki/CsrfToken-HowTo (unfortunately in German, but I hope that Google can help). Sorry, maybe it's not a big help, but there is no really a API documentation of fhem, only that site which I already mentioned. And I'm also still a beginner in fhem :/

enesbcs, 2019-05-29:
Google Translate definitely be needed.. this token-howto page seems good, as the examples shows simple HTTP GET requests with URL encoding. But ESPEasy sends JSON encapsulated data through HTTP POST calls in it's current _C009 controller plugin.

enesbcs, 2019-06-08:
It looks like that ESPEasy is using a dedicated ESPEasy Bridge in FHEM:
https://wiki.fhem.de/wiki/ESPEasy#ESPEasy_Bridge
I think i understood how it works, despite that i have zero german language knowledge. :)

enesbcs, 2019-06-08:
FHEM replies only with "WARNING: Wrong controller configured or ESPEasy Version is too old. WARNING: ESPEasy version R128 or later required." so this RPIEasy controller will not be ready for a while... :(

enesbcs, 2019-09-19:
I've downloaded the FHEM sources and i figured out that "module":"ESPEasy" is hardcoded in that... and Python's json library added extra spaces like " : " which failed on this test. OMG. Now i can proceed.

enesbcs, 2019-09-20:
Finally, from https://github.com/enesbcs/rpieasy/commit/27d8831f76a380098c88ac62cda96d97ce453bd6 RPIEasy FHEM controller is now available.

enesbcs, 2019-10-01:
No feedback provided case is closed as implemented.

closed 2019-10-01 by enesbcs