enesbcs/rpieasy #68: CMD via mqtt not working?

bennybubble, 2019-06-11:
Hi,

I tried to display a text on an OLED through MQTT using OPENHAB.
While I clearly see the command passing over MQTT, rpieasy doesn't seem to respond, also the log doesn't show this. I tried the same command via http and this does work!

`http://192.168.2.87/control?cmd=oled,0,0,Test` works fine
`19:58:00 CMD: oled,0,0,Test` in the rpieasy log file

Over MQTT I do the same and see this in the mosquitto_sub program:
`06/11/19 19:59:42.998 rpieasy087/oled/cmd 0,0,Test`

But no entry in the log file.

Any thoughts please ?
-ben

enesbcs, 2019-06-12:
If you read the plugin header it contains the solution:
#Commands can be remotely executed through MQTT with writing to:
#  topic: %sysname%/cmd with payload: command

I've tested and:
`mosquitto_pub -h MQTTSERVERIP -t 'rpieasy087/cmd' -m 'oled,0,0,Test'`
works as expected.

I am not sure what do you want with 'rpieasy087/oled/cmd' topic, it is simply not supported.

The changeable command topic is for the plugins:
If you leave the default "%sysname%/#/set" command topic than if you write for example '1' into 'systemname/taskname/valuename/set' topic, than the specified 'valuename' under the specific 'taskname' will be equal to 1. If you set command topic to "%sysname%/#" than every sent and received message will be examined as incoming data, which is not a good idea. (the /set behind it isolate them)
%tskname% and %valname% also supported, otherwise the "#" means %tskname%/%valname%

bennybubble, 2019-06-12:
Hi,

Thanks for your help, solved the problem by reading your example!

-ben

closed 2019-06-12 by bennybubble