enesbcs/rpieasy #69: Feature request: Implementation of LWT

bennybubble, 2019-06-12:
Hi,

Would it be possible to implement LWT like in ESPEasy ?

-ben

enesbcs, 2019-06-12:
It can be, however i've never used this feature.
I guess the controller plugin needs 3 parameters:
1: lwl topic
2: lwl connect message (unit sending it automatically to lwl topic when connected?)
3: lwl disconnect message (unit sending it automatically to lwl topic when disconnected?)

bennybubble, 2019-06-12:
Hi,

Yes correct, something like ESPEasy:

[image]

More information here [https://parg.co/4Jx](url)

TIA, Ben

enesbcs, 2019-06-13:
support added in commit https://github.com/enesbcs/rpieasy/commit/8832c754a9c016c2e3d514b6a35a50a0fa901318

bennybubble, 2019-06-14:
Thanks, updated my systems (2 RPI boards), tested them and it works perfectly!

closed 2019-06-14 by bennybubble