enesbcs/rpieasy #70: Feature Request: HT16K33 Keypad chip support.

Budman1758, 2019-06-18:
This chip supports quite a number of different keypads.  Support for it would be a pretty nice addition.  :)  It can also be used for LED applications but I don't have any experience doing that.

enesbcs, 2019-07-17:
The HTK16K33 chip is arrived, i have also bought a 8x8 LED matrix, but i have no clue how to wire them.. or either how to simulate keypad key pushing to it..

TD-er, 2019-07-17:
It is usually a matrix so you need to test all possibilities to see if one of them is shorted.

A while ago I did make some code to speed up this search, with as drawback that you can only detect 1 keypress (or use it to determine if you need to perform a more intensive check)
See my [code here](https://github.com/TD-er/ESPEasy/tree/bugfix/PCF8574_keypad)
It is not the same chip, but the principle should be the same.

Budman1758, 2019-07-17:
There is a breif explanation on wiring the keypad chip in the .ino file for ESPEasy HT16K33 keypad support. See the top of this link.  

[https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P058_HT16K33_KeyPad.ino](url)

That is pretty much how I wired up my keypad.

enesbcs, 2019-07-17:
I have read about this items a bit in the last hour, and as i understood i have to connect HT16K33 pins from A0..A7 (rows) to the LED matrix lower pins from the 1st pin (1 is marked) and C0..C7 (columns-cathode) to the upper 8 LED pins, based on this video tutorial and the HT16K33 datasheet:
http://arduino-er.blogspot.com/2015/01/how-to-identify-pin-1-of-8x8-led-matrix.html
I am afraid currently i only have this kind of keypad which may not good as column+row output is needed from the keypad:
[image]
Can i simulate a keypad with simple tactile switches and diodes somehow?

Budman1758, 2019-07-17:
> Can i simulate a keypad with simple tactile switches and diodes somehow to the HT16K33?

I see no reason why not.  All these keypads do is create a matrix of switches. Might be a bit complex to simulate with tact switches from a wiring standpoint but should work fine.

TD-er, 2019-07-17:
That 4 button keypad is also a matrix.
A 4x1 matrix :)

enesbcs, 2019-07-21:
HT16K33 support added in commit https://github.com/enesbcs/rpieasy/commit/a876e5ad4dc7d5009d8794ca2cd874b5c29ad84e

Budman1758, 2019-07-22:
Hooked up a keypad unit and all seems fine.  Pretty fast too.  Can't speak to the LED stuff tho....

enesbcs, 2019-07-23:
Thank you for your feedback. Keypresses queried 10 times per a sec, as i did not find any interrupt-like function. It is working in the same way as the ESPEasy P058 plugin.

closed 2019-07-24 by enesbcs