enesbcs/rpieasy #71: Feature request: PN533 RFID support.

Budman1758, 2019-06-18:
Another very common RFID chip.  Inexpensive too!  Have one working in my ESP keypad project and works quite well.

enesbcs, 2019-07-17:
The PN532 i've received behaves strangely: when i connect it to a nodemcu with ESPEasy, then i2scanner detects it once at 0x24 but after that the device disappears until power-cycling... 
And of course it is not working, only seeing this in the event log. Maybe it is a defective one?
450143: PN532: Read error: 214
450143: PN532: Reset on pin: 13
450421: PN532: BUS error
450421: PN532: Reset on pin: 13
450543: PN532: Read error: 215
450543: PN532: Reset on pin: 13
450821: PN532: BUS error

Budman1758, 2019-07-17:
Which particular module do you have?
Do you have the Reset pin connected? If I remember correctly you need to connect GPIO 13 to the reset pin on the PN532

enesbcs, 2019-07-17:
I have a red one, just in the picture on the letscontrolit wiki.. i've connected the reset to RSTPDN, without any positive change. Bus error is permanent, reset power-cycles the PN532 continously. (power led is blinking while resetting)
After the first I2C query, the PN532 pulls down the SCL to GND (and holding it down) which effectively kills I2C communication. Tried with attaching an I2C OLED to the same line, when the PN532 is attached, even the OLED is unavailable, when the PN532 is detached the OLED works.

Budman1758, 2019-07-17:
Strange.... Probably a dumb question but you don't have the I2C lines crossed up?

enesbcs, 2019-07-17:
Nope, if i crossing the lines, than the i2cscan finds absolutely nothing.

enesbcs, 2019-07-18:
I've tried it with 2.7k then 4.7k then 10k pullup resistors without success. :( Finally i'll give it up, asked a return from the seller and ordered another one from another seller. (registered airmail 20-31 days)

enesbcs, 2019-08-13:
The second one i got works without problem, support added in commit https://github.com/enesbcs/rpieasy/commit/226eb1f816b8b8fa36959615dbb1568dbd4ffe98

Budman1758, 2019-08-15:
Seems like it works fine as far as I can tell.  At first it seemed like the program kept finding and then not finding the PN532 module and was working very intermittently. However after a reboot of the PI it now seems pretty solid. I am running without the reset pin option.  So far no issues.

Nice work!!  :)

enesbcs, 2019-08-17:
Thank you for testing, feel free to reopen this case on related error.

closed 2019-08-17 by enesbcs