enesbcs/rpieasy #75: New interesting app

happytm, 2019-07-11:
I Came across this very new concept for easy IOT implementation.
https://github.com/linhtranvu/Quick-IoTs

I looks promising and developer promising linux version in future. But it looks like the easiest is the deployment through web browser as shown here:

https://quickiots.com/quick-iots-desktop/

Looks like it uses this backend gateway which in itself is very promising to implement different communication protocols in one bundle:

https://github.com/1technophile/OpenMQTTGateway

Sorry this is not in anyway related to RPIeasy issue but I posted it here just for discussion.

Thanks

enesbcs, 2019-07-11:
Quik-IoTs seems new, but already in archived state? Very interesting...
OpenMQTTGateway looks like a good project, i tried it before but failed to use my 433MHz weather station and with several 433Mhz bells, the same ones that is working with rtl_433 with RTLSDR and with the good old RCSwitch project so i am waiting OpenMQTTGateway to support a few more things in the future...

TD-er, 2019-07-12:
@enesbcs Maybe we should then discuss some things to keep interoperability between ESPeasy and RpiEasy as soon as you're starting implementing support for Lora.
I'm also working on adding support for TTN.

enesbcs, 2019-07-12:
@TD-er Sure, interoperability would be nice. I am still waiting the SX1276 for testing purposes, but i will try to use this library https://pypi.org/project/pyLoRa/ without encryption. But it is only the network layer, we have to figure out some kind of headers which is small enough, but make it possible some kind of addressing between the units. I think that the 4-5 bytes simple header based on Libelium ( see: http://cpham.perso.univ-pau.fr/LORA/RPIgateway.html ) can be used or can be combined with the current ESPEasy P2P UDP package types, that will also make the decoding process easier. The big question is: 254 nodes ought to be enough for anybody?

TD-er, 2019-07-12:
The bad thing about the SX127x units is that you need to run the entire LoraWan stack including all time-critical stuff.
I am now testing the Microchip RN2483, which is just a UART interface and all the LoraWan stack is included on the unit itself.
Make sure you have the latest version of the module (version relates partly to hardware, but mostly to software revision)

enesbcs, 2019-07-12:
"LoraWan" is bigger indeed, but i targeted only simple "Lora" without the heavy stuff.

TD-er, 2019-07-12:
That still involves quite time critical interaction with the SX127x.

closed 2019-08-21 by happytm