enesbcs/rpieasy #77: OLED clearline not working?

bennybubble, 2019-07-14:
Hi,

Im using an oled (identified as PCF8574A/SSD1305/SSD1306 monochrome OLED when scanning I2C) to display some text from OpenHab using MQTT.
The text is written to the display (so communication is okay) but when using clearline, the line is not erased!
[image]

Command used to write text : 
    rpi047_Cmd.sendCommand("oled,1,0,Alarm ON")
Command used to erase line : 
    rpi047_Cmd.sendCommand("oledcmd,clearline,1")

Any idea of the reason for this?

enesbcs, 2019-07-15:
You are right it is a bug, but not in clearline, but in the oled write command which is not meant to be 0 based... so "oled,1,0" wrote the wrong line. It is now fixed in commit https://github.com/enesbcs/rpieasy/commit/a95e7710be59b5b9b056ad30794425a3b4210d93

bennybubble, 2019-07-16:
Tnx, tested and working okay!

closed 2019-07-16 by bennybubble