enesbcs/rpieasy #78: Nice little tool for RPIEasy users to make installation even more easier

happytm, 2019-07-14:
Recently I had to install Raspbian headless setup for about 20 RPI Zero W so I was looking to make my life easier and came across PIBakery software which I very much liked so for benefit of RPIEasy users who did not know about it following is the link for the software:

https://www.pibakery.org/

Thanks

enesbcs, 2019-07-15:
Thank you for this tip, added to the wiki!

closed 2019-07-15 by enesbcs

ironataerial, 2019-07-15:
Hello,

May I ask how did you go about acquiring 20pcs RPI zero when most vendors
sell them in singles ?

-D

happytm, 2019-07-15:
I bought it at Micro Center store in Boston over period of time. And it cost me only $5 each.See below.

https://www.microcenter.com/product/486575/zero-w

Thanks