enesbcs/rpieasy #81: Show JSON button under tools

happytm, 2019-07-27:
Looks like Show JSON button under tools menu in UI is not responding.

Thanks

enesbcs, 2019-07-28:
I've just tested it's working for me with 2 enabled Protocols (ESPEasy P2P&Domoticz MQTT) with the following 13 used devices: DHT11, BH1750, Output helper, Run OS command, System Info, Pygame Sound Player, VLC Sound PLayer, Dual Switch device, Input Switch device, Dummy device, Pygame Sound Background player, Input Evdev.

Please send me your configuration to be able to debug.

Budman1758, 2019-07-28:
I have 2 different units running. One using build  0.19.176  and the other using build  0.19.208  and both of them are working as they should.

happytm, 2019-07-28:
OK it is working now. I think I had build 0.19.207 running but instead of updating whole firmware I just replaced the commands.py file on pi. Now I notice update button under tools which makes it so easy. Once I updated to latest build it is working fine.

Sorry for the trouble.

Thanks

enesbcs, 2019-07-29:
Thanks for feedback, it was a bug that had been fixed before. :)

closed 2019-07-29 by enesbcs