enesbcs/rpieasy #82: Feature Request : Data logger

happytm, 2019-07-30:
Since RPIEasy is now very mature and almost covers most of the common sensor plugins we should think about automatic data logging feature. Espeasy has data storage limitation which RPIEasy does not have. I use SQlite database to insert all values (total of 198 columns) from json output of RPIEasy using node-red. There are few options for time series databases but to start with SQlite seems easier to use. I have couple of links below:

http://www.sqlitetutorial.net/sqlite-python/

The one I use is using node-red:

https://www.youtube.com/watch?v=nkKf26oKzhQ

Thanks

enesbcs, 2019-07-30:
SQLite is a very good idea, i used it a long time before, when i did some programming in Lazarus. However you have to consider that Raspberry Pi wear out SD cards after some time, so data logging into the same SD card where the system is running from is not a generally good idea. Maybe logging to an external USB drive will work OK.
How exactly you think this data storage implemantation in RPIEasy system? I see several ways for example 1: implementing SQLite as a Controller, and on every device plugin it can be selected whether it store values onto it or not
2: can be implemented into RPIEasy core as paralell logging type next to serial-console-syslog-weblog in this way it logs every log line into SQLite
3: can be implemented as a Notification plugin, and a value can be added into it from Rules only

Budman1758, 2019-07-30:
This would be a great addition to the program.  Please consider integrating MYSql support also. I use MYSql for several different databases here and to add to it would be awesome.

TD-er, 2019-07-30:
You can also look into SQLAlchemy.
That's a Python wrapper for a lot of SQL databases.
I recently made an interface in Python using SQLAlchemy and it works seemingly between MSSQL (in Azure) and Postgres.
I also have some Flask framework around it which implements the REST API for it to store the data.

Currently I am working to integrate the ESPeasy cache controller to store data into this database.
Dumping it into a CSV file from ESPeasy and import that via Python is already working and live updates also (just a simple HTTP GET and POST interface)
But I also would like to 'dump' the raw binary files to a server when there's connection and import that into the database.

Maybe we can somewhat integrate this to support it on both ESPeasy and RpiEasy?

happytm, 2019-07-30:
I agree with option 1 too. That way the user have option to activate data logging easily.Right now I use all json data coming in from other devices by using http://myip/json so it can be used later on however it is used to display data.If implemented it should have option of receiving datasets via json from all other ESPEasy devices on network and dumped into same table so it can be filtered out from that big table to display selective data.Following is my data table creation and data entry commands:

Table creation:

`CREATE TABLE 'bigdata' ('id' INTEGER PRIMARY KEY AUTOINCREMENT NOT NULL,'time' INTEGER,'unit' INTEGER,'build' TEXT,'plugins' INTEGER,'uptime' INTEGER,'load'TEXT,'memory' INTEGER,'rssi' TEXT,'ssid' TEXT,'ipconfig' TEXT,'ip' INTEGER,'subnet' INTEGER,'gateway' INTEGER,'mac' TEXT,'dns' INTEGER,'device' TEXT,'taskno1' INTEGER,'taskno2' INTEGER,'taskno3' INTEGER,'taskno4' INTEGER,'taskno5' INTEGER,'taskno6' INTEGER,'taskno7' INTEGER,'taskno8' INTEGER,'taskno9' INTEGER,'taskno10' INTEGER,'taskno11' INTEGER,'taskno12' INTEGER,'taskname1' TEXT,'taskname2' TEXT,'taskname3' TEXT,'taskname4' TEXT,'taskname5' TEXT,'taskname6' TEXT,'taskname7' TEXT,'taskname8' TEXT,'taskname9' TEXT,'taskname10' TEXT,'taskname11' TEXT,'taskname12' TEXT,'enabled1' TEXT,'enabled2' TEXT,'enabled3' TEXT,'enabled4' TEXT,'enabled5' TEXT,'enabled6' TEXT,'enabled7' TEXT,'enabled8' TEXT,'enabled9' TEXT,'enabled10' TEXT,'enabled11' TEXT,'enabled12' TEXT,'valueno11' INTEGER,'valueno12' INTEGER,'valueno13' INTEGER,'valueno14' INTEGER,'valueno21' INTEGER,'valueno22' INTEGER,'valueno23' INTEGER,'valueno24' INTEGER,'valueno31' INTEGER,'valueno32' INTEGER,'valueno33' INTEGER,'valueno34' INTEGER,'valueno41' INTEGER,'valueno42' INTEGER,'valueno43' INTEGER,'valueno44' INTEGER,'valueno51' INTEGER,'valueno52' INTEGER,'valueno53' INTEGER,'valueno54' INTEGER,'valueno61' INTEGER,'valueno62' INTEGER,'valueno63' INTEGER,'valueno64' INTEGER,'valueno71' INTEGER,'valueno72' INTEGER,'valueno73' INTEGER,'valueno74' INTEGER,'valueno81' INTEGER,'valueno82' INTEGER,'valueno83' INTEGER,'valueno84' INTEGER,'valueno91' INTEGER,'valueno92' INTEGER,'valueno93' INTEGER,'valueno94' INTEGER,'valueno101' INTEGER,'valueno102' INTEGER,'valueno103' INTEGER,'valueno104' INTEGER,'valueno111' INTEGER,'valueno112' INTEGER,'valueno113' INTEGER,'valueno114' INTEGER,'valueno121' INTEGER,'valueno122' INTEGER,'valueno123' INTEGER,'valueno124' INTEGER,'sensorname11' TEXT,'sensorname12' TEXT,'sensorname13' TEXT,'sensorname14' TEXT,'sensorname21' TEXT,'sensorname22' TEXT,'sensorname23' TEXT,'sensorname24' TEXT,'sensorname31' TEXT,'sensorname32' TEXT,'sensorname33' TEXT,'sensorname34' TEXT,'sensorname41' TEXT,'sensorname42' TEXT,'sensorname43' TEXT,'sensorname44' TEXT,'sensorname51' TEXT,'sensorname52' TEXT,'sensorname53' TEXT,'sensorname54' TEXT,'sensorname61' TEXT,'sensorname62' TEXT,'sensorname63' TEXT,'sensorname64' TEXT,'sensorname71' TEXT,'sensorname72' TEXT,'sensorname73' TEXT,'sensorname74' TEXT,'sensorname81' TEXT,'sensorname82' TEXT,'sensorname83' TEXT,'sensorname84' TEXT,'sensorname91' TEXT,'sensorname92' TEXT,'sensorname93' TEXT,'sensorname94' TEXT,'sensorname101' TEXT,'sensorname102' TEXT,'sensorname103' TEXT,'sensorname104' TEXT,'sensorname111' TEXT,'sensorname112' TEXT,'sensorname113' TEXT,'sensorname114' TEXT,'sensorname121' TEXT,'sensorname122' TEXT,'sensorname123' TEXT,'sensorname124' TEXT,'value11' REAL,'value12'REAL,'value13' REAL,'value14' REAL,'value21' REAL,'value22' REAL,'value23' REAL,'value24' REAL,'value31' REAL,'value32' REAL,'value33' REAL,'value34' REAL,'value41' REAL,'value42' REAL,'value43' REAL,'value44' REAL,'value51' REAL,'value52' REAL,'value53' REAL,'value54' REAL,'value61' REAL,'value62' REAL,'value63' REAL,'value64' REAL,'value71' REAL,'value72' REAL,'value73' REAL,'value74' REAL,'value81' REAL,'value82' REAL,'value83' REAL,'value84' REAL,'value91' REAL,'value92' REAL,'value93' REAL,'value94' REAL,'value101' REAL,'value102' REAL,'value103' REAL,'value104' REAL,'value111' REAL,'value112' REAL,'value113' REAL,'value114' REAL,'value121' REAL,'value122' REAL,'value123' REAL,'value124' REAL,'epoch' INTEGER)`

To insert data:

`INSERT INTO bigdata (id,time,unit,build,plugins,uptime,load,memory,rssi,ssid,ipconfig,ip,subnet,gateway,mac,dns,device,taskno1,taskno2,taskno3,taskno4,taskno5,taskno6,taskno7,taskno8,taskno9,taskno10,taskno11,taskno12,taskname1,taskname2,taskname3,taskname4,taskname5,taskname6,taskname7,taskname8,taskname9,taskname10,taskname11,taskname12,enabled1,enabled2,enabled3,enabled4,enabled5,enabled6,enabled7,enabled8,enabled9,enabled10,enabled11,enabled12,valueno11,valueno12,valueno13,valueno1valueno21,valueno22,valueno23,valueno24,valueno31,valueno32,valueno33,valueno34,valueno41,valueno42,valueno43,valueno44,valueno51,valueno52,valueno53,valueno54,valueno61,valueno62,valueno63,valueno64,valueno71,valueno72,valueno73,valueno74,valueno81,valueno82,valueno83,valueno84,valueno91,valueno92,valueno93,valueno94,valueno101,valueno102,valueno103,valueno104,valueno111,valueno112,valueno113,valueno114,valueno121,valueno122,valueno123,valueno124,sensorname11,sensorname12,sensorname13,sensorname14,sensorname21,sensorname22,sensorname23,sensorname24,sensorname31,sensorname32,sensorname33,sensorname34,sensorname41,sensorname42,sensorname43,sensorname44,sensorname51,sensorname52,sensorname53,sensorname54,sensorname61,sensorname62,sensorname63,sensorname64,sensorname71,sensorname72,sensorname73,sensorname74,sensorname81,sensorname82,senso [... cut ...]
"{{payload.Sensors[9].TaskValues[0].Value}}","{{payload.Sensors[9].TaskValues[1].Value}}","{{payload.Sensors[9].TaskValues[2].Value}}","{{payload.Sensors[9].TaskValues[3].Value}}",
"{{payload.Sensors[10].TaskValues[0].Value}}","{{payload.Sensors[10].TaskValues[1].Value}}","{{payload.Sensors[10].TaskValues[2].Value}}","{{payload.Sensors[10].TaskValues[3].Value}}",
`"{{payload.Sensors[11].TaskValues[0].Value}}","{{payload.Sensors[11].TaskValues[1].Value}}","{{payload.Sensors[11].TaskValues[2].Value}}","{{payload.Sensors[11].TaskValues[3].Value}}");`

happytm, 2019-07-30:
Anyone using Node-red these are the nodes I use for SQLite database:

`[{"id":"7c9b9844.a19268","type":"debug","z":"b6187ee2.d3e4a","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":510,"y":1740,"wires":[]},{"id":"ee9fe47.2360218","type":"sqlite","z":"b6187ee2.d3e4a","mydb":"4e196809.c9f1d8","sqlquery":"msg.topic","sql":"","name":"Table Data In","x":350,"y":1780,"wires":[["201f2277.6f7b7e"]]},{"id":"15b11cd0.70d6c3","type":"inject","z":"b6187ee2.d3e4a","name":"CREATE DB","topic":"CREATE TABLE 'bigdata' ('id' INTEGER PRIMARY KEY AUTOINCREMENT NOT NULL,'time' INTEGER,'unit' INTEGER,'build' TEXT,'plugins' INTEGER,'uptime' INTEGER,'load'TEXT,'memory' INTEGER,'rssi' TEXT,'ssid' TEXT,'ipconfig' TEXT,'ip' INTEGER,'subnet' INTEGER,'gateway' INTEGER,'mac' TEXT,'dns' INTEGER,'device' TEXT,'taskno1' INTEGER,'taskno2' INTEGER,'taskno3' INTEGER,'taskno4' INTEGER,'taskno5' INTEGER,'taskno6' INTEGER,'taskno7' INTEGER,'taskno8' INTEGER,'taskno9' INTEGER,'taskno10' INTEGER,'taskno11' INTEGER,'taskno12' INTEGER,'taskname1' TEXT,'taskname2' TEXT,'taskname3' TEXT,'taskname4' TEXT,'taskname5' TEXT,'taskname6' TEXT,'taskname7' TEXT,'taskname8' TEXT,'taskname9' TEXT,'taskname10' TEXT,'taskname11' TEXT,'taskname12' TEXT,'enabled1' TEXT,'enabled2' TEXT,'enabled3' TEXT,'enabled4' TEXT,'enabled5' TEXT,'enabled6' TEXT,'enabled7' TEXT,'enabled8' TEXT,'enabled9' TEXT,'enabled10' TEXT,'enabled11' TEXT,'enabled12' TEXT,'valueno11' INTEGER,'valueno12' INTEGER,'valueno13' INTEGER,'valueno14' INTEGER,'valueno21' INTEGER,'valueno22' INTEGER,'valueno23' INTEGER,'valueno24' INTEGER,'valueno31' INTEGER,'valueno32' INTEGER,'valueno33' INTEGER,'valueno34' INTEGER,'valueno41' INTEGER,'valueno42' INTEGER,'valueno43' INTEGER,'valueno44' INTEGER,'valueno51' INTEGER,'valueno52' INTEGER,'valueno53' INTEGER,'valueno54' INTEGER,'valueno61' INTEGER,'valueno62' INTEGER,'valueno63' INTEGER,'valueno64' INTEGER,'valueno71' INTEGER,'valueno72' INTEGER,'valueno73' INTEGER,'valueno74' INTEGER,'valueno81' INTEGER,'valueno82' INTEGER,'valueno83' INTEGER,'valueno84' INTEGER,'valueno91' INTEGER,'valueno92' INTEGER,'valueno93' INTEGER,'valueno94' INTEGER,'valueno101' INTEGER,'valueno102' INTEGER,'valueno103' INTEGER,'valueno104' INTEGER,'valueno111' INTEGER,'valueno112' INTEGER,'valueno113' INTEGER,'valueno114' INTEGER,'valueno121' INTEGER,'valueno122' INTEGER,'valueno123' INTEGER,'valueno124' INTEGER,'sensorname11' TEXT,'sensorname12' TEXT,'sensorname13' TEXT,'sensorname14' TEXT,'sensorname21' TEXT,'sensorname22' TEXT,'sensorname23' TEXT,'sensorname24' TEXT,'sensorname31' TEXT,'sensorname32' TEXT,'sensorname33' TEXT,'sensorname34' TEXT,'sensorname41' TEXT,'sensorname42' TEXT,'sensorname43' TEXT,'sensorname44' TEXT,'sensorname51' TEXT,'sensorname52' TEXT,'sensorname53' TEXT,'sensorname54' TEXT,'sensorname61' TEXT,'sensorname62' TEXT,'sensorname63' TEXT,'sensorname64' TEXT,'sensorname71' TEXT,'sensorname72' TEXT,'sensorname73' TEXT,'sensorname74' TEXT,'sensorname81' TEXT,'sensorname82' TEXT,'sensorname83' TEXT,'sensorname84' TEXT,'sensorname91' TEXT,'sensorname92' TEXT,'sensorname93' TEXT,'sensorname94' TEXT,'sensorname101' TEXT,'sensorname102' TEXT,'sensorname103' TEXT,'sensorname104' TEXT,'sensorname111' TEXT,'sensorname112' TEXT,'sensorname113' TEXT,'sensorname114' TEXT,'sensorname121' TEXT,'sensorname122' TEXT,'sensorname123' TEXT,'sensorname124' TEXT,'value11' REAL,'value12'REAL,'value13' REAL,'value14' REAL,'value21' REAL,'value22' REAL,'value23' REAL,'value24' REAL,'value31' REAL,'value32' REAL,'value33' REAL,'value34' REAL,'value41' REAL,'value42' REAL,'value43' REAL,'value44' REAL,'value51' REAL,'value52' REAL,'value53' REAL,'value54' REAL,'value61' REAL,'value62' REAL,'value63' REAL,'value64' REAL,'value71' REAL,'value72' REAL,'value73' REAL,'value74' REAL,'value81' REAL,'value82' REAL,'value83' REAL,'value84' REAL,'value91' REAL,'value92' REAL,'value93' REAL,'value94' REAL,'value101' REAL,'value102' REAL,'value103' REAL,'value104' REAL,'value111' REAL,'value112' REAL,'value113' REAL,'value114' REAL,'value121' REAL,'value122' REAL,'value123' REAL,'value124' REAL,'epoch' INTEGER)","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":150,"y":1780,"wires":[["ee9fe47.2360218"]]},{"id":"201f2277.6f7b7e","type":"debug","z":"b6187ee2.d3e4a","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":510,"y":1780,"wires":[]},{"id":"b2db970f.388b38","type":"template","z":"b6187ee2.d3e4a","name":"Insert To bigdata","field":"topic","fieldType":"msg","format":"handlebars","syntax":"mustache","template":"INSERT INTO bigdata (time,unit,build,plugins,uptime,load,memory,rssi,ssid,ipconfig,ip,subnet,gateway,mac,dns,device,taskno1,taskno2,taskno3,taskno4,taskno5,taskno6,taskno7,taskno8,taskno9,taskno10,taskno11,taskno12,taskname1,taskname2,taskname3,taskname4,taskname5,taskname6,taskname7,taskname8,taskname9,taskname10,taskname11,taskname12,enabled1,enabled2,enabled3,enabled4,enabled5,enabled6,enabled7,enabled8,enabled9,enabled10,enabled11,enabled12,valueno11,valueno12,valueno13,valueno14,valueno21,valueno22,valueno23,valueno24,valueno31,valueno32,valueno33,valueno34,valueno41,valueno42,valueno43,valueno44,valueno51,valueno52,valueno53,valueno54,valueno61,valueno62,valueno63,valueno64,valueno71,valueno72,valueno73,valueno74,valueno81,valueno82,valueno83,valueno84,valueno91,valueno92,valueno93,valueno94,valueno101,valueno102,valueno103,valueno104,va [... cut ...] inject","z":"b6187ee2.d3e4a","name":"Drop DB","topic":"DROP TABLE bigdata;","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":162,"y":1835,"wires":[["e7b52818.7782f8"]]},{"id":"f512801a.872a","type":"debug","z":"b6187ee2.d3e4a","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":518,"y":1831,"wires":[]},{"id":"4e196809.c9f1d8","type":"sqlitedb","z":"","db":"/home/pi/rpieasy/data/bigdata","mode":"RWC"}]`

Thanks

happytm, 2019-08-01:
Just for reference:

https://www.youtube.com/watch?v=cYWiDiIUxQc

Thanks

enesbcs, 2019-08-02:
Experimental SQLite and MySQL support added in C016 controller into RPIEasy 19.214.
Data structure may change in the future if needed, however i do not really think that a very large number of column is generally good.

Budman1758, 2019-08-02:
I set up a unit to use MYSQL and seems to work pretty well. Created the database and started populating it without any errors that I can see.  Well done!!

happytm, 2019-08-03:
Wow! that was fast.Thank you for implementing database engine for RPIEasy. Day by day it is living up to it's name being easy.I was logging & displaying data on node-red dashboard within couple of minutes. Please look at the data displayed below. The date structure looks little too long and I did not understand Sensor field but other than that it is flowless:

[image]

Following is a simple flow I used to display data in a table:

`[{"id":"a5b3a162.ff3de","type":"inject","z":"b6187ee2.d3e4a","name":"Display Data","topic":"SELECT * FROM easysensor ORDER BY time  DESC LIMIT 10;","payload":"","payloadType":"date","repeat":"60","crontab":"","once":false,"onceDelay":0.1,"x":120,"y":1120,"wires":[["ffff5ef.ecf09a"]]},{"id":"ffff5ef.ecf09a","type":"sqlite","z":"b6187ee2.d3e4a","mydb":"3d6e513f.0a04ee","sqlquery":"msg.topic","sql":"","name":"Data Out","x":300,"y":1120,"wires":[["516b55f6.fd3d5c"]]},{"id":"516b55f6.fd3d5c","type":"ui_template","z":"b6187ee2.d3e4a","group":"632d7263.26704c","name":"","order":1,"width":"20","height":"8","format":"\n<link rel=\"stylesheet\" href=\"https://stackpath.bootstrapcdn.com/bootstrap/4.3.1/css/bootstrap.min.css\" integrity=\"sha384-ggOyR0iXCbMQv3Xipma34MD+dH/1fQ784/j6cY/iJTQUOhcWr7x9JvoRxT2MZw1T\" crossorigin=\"anonymous\">\n<script src=\"https://code.jquery.com/jquery-3.3.1.slim.min.js\" integrity=\"sha384-q8i/X+965DzO0rT7abK41JStQIAqVgRVzpbzo5smXKp4YfRvH+8abtTE1Pi6jizo\" crossorigin=\"anonymous\"></script>\n<script src=\"https://cdnjs.cloudflare.com/ajax/libs/popper.js/1.14.7/umd/popper.min.js\" integrity=\"sha384-UO2eT0CpHqdSJQ6hJty5KVphtPhzWj9WO1clHTMGa3JDZwrnQq4sF86dIHNDz0W1\" crossorigin=\"anonymous\"></script>\n<script src=\"https://stackpath.bootstrapcdn.com/bootstrap/4.3.1/js/bootstrap.min.js\" integrity=\"sha384-JjSmVgyd0p3pXB1rRibZUAYoIIy6OrQ6VrjIEaFf/nJGzIxFDsf4x0xIM+B07jRM\" crossorigin=\"anonymous\"></script>\n <script src=\"https://ajax.googleapis.com/ajax/libs/angularjs/1.7.8/angular.min.js\"></script>\n<div>\n  \n     <form>\n      <span class=\"input-group\">\n          <i class=\"fa fa-search\"></i>\n        <input type=\"text\" class=\"form-control\" placeholder=\"Search\" ng-model=\"search\">\n      </span>      \n  </form>\n  <div class=\"container\" ng-app=\"sortApp\">\n<table style=\"width:100%\">\n      \n        <thead>\n        <tr style=\"width:100%\">\n            <td>\n          <a href=\"#\">\n            Index\n          </a>\n        </td>\n        \n        <td>\n          <a href=\"#\" ng-click=\"sortType = 'time'; sortReverse = !sortReverse\">\n            Time\n            <span ng-show=\"sortType == 'time' && !sortReverse\" class=\"fa fa-caret-down\"></span>\n            <span ng-show=\"sortType == 'time' && sortReverse\" class=\"fa fa-caret-up\"></span>\n          </a>\n        </td>\n        <td>\n          <a href=\"#\" ng-click=\"sortType = 'device'; sortReverse = !sortReverse\">\n            Unit\n            <span ng-show=\"sortType == 'device' && !sortReverse\" class=\"fa fa-caret-down\"></span>\n            <span ng-show=\"sortType == 'device' && sortReverse\" class=\"fa fa-caret-up\"></span>\n          </a>\n        </td>\n\t\t\n        \n\t\t\n\t\t<td>\n          <a href=\"#\" ng-click=\"sortType = 'nodename'; sortReverse = !sortReverse\">\n            Device\n            <span ng-show=\"sortType == 'nodename' && !sortReverse\" class=\"fa fa-caret-down\"></span>\n            <span ng-show=\"sortType == 'nodename' && sortReverse\" class=\"fa fa-caret-up\"></span>\n          </a>\n        </td>\n\t\t\n\t\t\n\t\t\n        <td>\n          <a href=\"#\" ng-click=\"sortType = 'rssi'; sortReverse = !sortReverse\">\n          RSSI\n            <span ng-show=\"sortType == 'rssi' && !sortReverse\" class=\"fa fa-caret-down\"></span>\n            <span ng-show=\"sortType == 'rssi' && sortReverse\" class=\"fa fa-caret-up\"></span>\n          </a>\n        </td>\n\t\t<td>\n          <a href=\"#\" ng-click=\"sortType = 'battery'; sortReverse = !sortReverse\">\n            Battery\n            <span ng-show=\"sortType == 'battery' && !sortReverse\" class=\"fa fa-caret-down\"></span>\n            <span ng-show=\"sortType == 'battery' && sortReverse\" class=\"fa fa-caret-up\"></span>\n          </a>\n        </td>\n\t\t\n\t\t<td>\n          <a href=\"#\" ng-click=\"sortType = 'tasknum'; sortReverse = !sortReverse\">\n            Task No\n            <span ng-show=\"sortType == 'tasknum' && !sortReverse\" class=\"fa fa-caret-down\"></span>\n            <span ng-show=\"sortType == 'tasknum' && sortReverse\" class=\"fa fa-caret-up\"></span>\n          </a>\n        </td>\n\t\t\n\t\t<td>\n          <a href=\"#\" ng-click=\"sortType = 'taskname'; sortReverse = !sortReverse\">\n            Task Name\n            <span ng-show=\"sortType == 'taskname' && !sortReverse\" class=\"fa fa-caret-down\"></span>\n            <span ng-show=\"sortType == 'taskname' && sortReverse\" class=\"fa fa-caret-up\"></span>\n          </a>\n        </td>\n\t\t\n\t\t\n        <td>\n          <a href=\"#\" ng-click=\"sortType = '(sensortype -0)'; sortReverse = !sortReverse\">\n            Sensor\n            <span ng-show=\"sortType == '(sensortype -0)' && !sortReverse\" class=\"fa fa-caret-down\"></span>\n            <span ng-show=\"sortType == '(sensortype -0)' && sortReverse\" class=\"fa fa-caret-up\"></span>\n          </a>\n        </td>\n        \n         <td>\n          <a href=\"#\" ng-click=\"sortType = '(value1 -0)'; sortReverse = !sortReverse\">\n          [... cut ...] ","storeOutMessages":true,"fwdInMessages":true,"templateScope":"local","x":480,"y":1120,"wires":[["bfb13314.1ddb7"]]},{"id":"3d6e513f.0a04ee","type":"sqlitedb","z":"","db":"/home/pi/rpieasy/easydata","mode":"RWC"},{"id":"632d7263.26704c","type":"ui_group","z":"","name":"SensorData","tab":"62332cd.61a90d4","order":2,"disp":true,"width":"20","collapse":true},{"id":"62332cd.61a90d4","type":"ui_tab","z":"","name":"Status","icon":"dashboard","order":3,"disabled":false,"hidden":false}]`

Thanks again.

enesbcs, 2019-08-03:
Yes, for SQLite, the "unit", "nodename" is not containing much information, but for a central MySQL they can be quite useful. :) The sensor column is simply contains the sensor type, which gives informations about the values, it may or may not be useful..
SENSOR_TYPE_NONE                 =  0
SENSOR_TYPE_SINGLE               =  1
SENSOR_TYPE_TEMP_HUM             =  2
SENSOR_TYPE_TEMP_BARO            =  3
SENSOR_TYPE_TEMP_HUM_BARO        =  4
SENSOR_TYPE_DUAL                 =  5
SENSOR_TYPE_TRIPLE               =  6
SENSOR_TYPE_QUAD                 =  7
SENSOR_TYPE_TEMP_EMPTY_BARO      =  8
SENSOR_TYPE_SWITCH               = 10
SENSOR_TYPE_DIMMER               = 11
SENSOR_TYPE_LONG                 = 20
SENSOR_TYPE_WIND                 = 21
SENSOR_TYPE_TEXT                 = 101

enesbcs, 2019-08-03:
The "time" column is a simple INTEGER type, it can be formatted by the SQL query,  and contains the UNIX time. http://www.sqlitetutorial.net/sqlite-date/

enesbcs, 2019-08-12:
Issue closed as no problems reported. Feel free to reopen in case of related errors.

closed 2019-08-12 by enesbcs