enesbcs/rpieasy #90: P1 meter

Sharkymarc, 2019-09-25:
Is it possible to get P044 P1 wifi gateway working in rpi easy? Or even a P1 smart meter usb kabel to work in rpi easy?

enesbcs, 2019-09-25:
I guess it is possible, as it looks like a simple serial device, but unfortunately i do not have one, so i do not think that i will do it in the near future.

Sharkymarc, 2019-09-25:
Dear Alexander,

In the Netherlands their is guy who implemented the code into espeasy. I will ask him to contribute the code to rpieasy. 

Outlook voor Android downloaden

TD-er, 2019-09-25:
The signal on the P1 bus is inverted.
So either you have to electrical invert the signal, or set the UART port so that it can process inverted signals.
Not sure if the R'pi can do that.

Sharkymarc, 2019-09-26:
Dear Gijs,

I wonder if rpieasy can readout the input of an P1 usb smart meter kabel. That can be inserted into a rpi.

Marc Wulms 

Outlook voor Android downloaden

enesbcs, 2020-03-13:
Closed as no activity in the last 6 months.

closed 2020-03-13 by enesbcs