enesbcs/rpieasy #91: Missing Controller: Generic HTTP

llevet, 2019-09-28:
Hi,
It miss controller:  Generic http.
it is useful to use it with Jeedom.

Thank's.

Ludo.

enesbcs, 2019-09-28:
Hi,
You mean C008 controller?
https://github.com/letscontrolit/ESPEasy/blob/mega/src/_C008.ino

llevet, 2019-09-29:
Hi Alexander,

Yes, exactely.

Ludo.

enesbcs, 2019-09-29:
C008 controller for simple GET HTTP request is now implemented. Authentication is not yet supported,  %sysname%, %tskname%, %valname% and %value% is supported like the original.

llevet, 2019-09-29:
Hi,
Perfect ! It is working ! 

Please, normaly for jeedom (plugin espeasy) the url for Controller Publish is : 
device=%sysname%&taskid=%id%&cmd=%valname%&value=%value%

We use taskid number (%id%) and not  %tskname%.
I put it to 0 like this (i see on my esp8266 that %id% return 0 or 1 but alway the same value) :  
device=%sysname%&taskid=0&cmd=%valname%&value=%value%
and it work well. I don't know if it is important or not, but it work like that.

Thank's a lot.

Ludo.

enesbcs, 2019-09-29:
Sure, I can add %id% that will be the number of the task 1..48.

llevet, 2019-09-29:
Hi,
Perfect it work.
Thank's.

Ludo.

closed 2019-09-29 by llevet