enesbcs/rpieasy #92: Password interface ask password for control

llevet, 2019-09-29:
Hi,
When i protect by Admin Password the interface of RPIEasy, it protect too the call to control url like this : http://x.x.x.x/control?cmd=GPIO,17,0

But in espeasy it isn't. Control url is always without password.

Please could you add in 'Config tab' a checkbox to disable password for control url.
Like this, we can protect configuration but control url are protected or not by password.

Thanks.

Ludo.

enesbcs, 2019-09-29:
Can be added but it is a serious security breach, if anyone can execute reset command remotely then password protection for webgui is almost superfluous. But of course it can be implemented.

llevet, 2019-09-29:
Hi,
Yes, but i haven't tested if all the commands with espeasy are usable without password.
You can authorize only no dangerous command like GPIO,Pulse,LongPulse,PWM,Status   ... 
So the label of the checkbox will be :
Disable password for safe command (GPIO,Pulse,LongPulse,PWM,Status...)

Ludo.

enesbcs, 2019-09-30:
Implemented RPIEasy core commands are: delay,taskrun,taskvalueset,taskvaluesetandrun,timerpause,timerset,event,sendto,publish,sendtohttp,sendtoudp,reboot,reset,halt,update,exit,notify,setvolume

Controller related commands:
blecommand,loracommand,espnowcommand,serialcommand

And there are some external commands, which are implemented by plugins:

P001&P029:
gpio,pwm,pulse,longpulse,

P007:
pfcdac

P009:
mcpgpio,mcppulse,mcplongpulse

P011:
extgpio,extpwm,extpulse,extlongpulse

P012:
lcdcmd,lcd

P019:
pcfgpio,pcfpulse,pcflongpulse

P022:
pcapwm,pcafrq

P023:
oledcmd,oled

P036:
oledframedcmd

P038:
neopixel, neopixelall,neopixelline

P049:
mhzcalibratezero, mhzreset, mhzabcenable, mhzabcdisable, mhzmeasurementrange2000, mhzmeasurementrange5000

P057:
M,MNUM,MBR

P073:
7don, 7diff, 7db

P112:
rfsend,rfsenddec,rfprotocol,rfrepeat

P201:
serialwrite

P203:
dac

P204:
motor

P501:
usbrelay

P502:
playaudio

P503:
playaudiobg

P505:
vlcaudio

Every command is documented in their plugin source header.

llevet, 2019-09-30:
Ok like this : 
Disable password for safe command (EXCLUDE: reboot,reset,halt ...)

Please add your last message to WIKI, it is interesting for all users !

Thanks.

Ludo.

enesbcs, 2019-10-01:
Option added in commit https://github.com/enesbcs/rpieasy/commit/286caa7acdc30dd7bef931e50ae97d42c92269bd

llevet, 2019-10-01:
Nice ! It work for me !

Ludo.

closed 2019-10-01 by llevet