enesbcs/rpieasy #94: Some way to keep plugin IDs in sync with ESPeasy

TD-er, 2019-09-30:
First of all, I want to say "wow" to the immense speed you manage to keep when building Rpi-Easy from scratch.

I've noticed you did keep almost all of the existing ESPeasy plugin IDs in sync in Rpi easy.
Since there will in the end more interaction between the platforms (e.g. via p2p or LoRa(WAN)), I think it would be nice to have some shared list to make sure new plugins either of us will add won't conflict in their numbers.

For example, we can keep some list in which a number is claimed when either of us is adding a new plugin. 
The question remains, where to keep such a list.
In a separate file in the ESPeasy repository? (e.g. in the documentation, so we can link to eachother)

enesbcs, 2019-09-30:
For me is OK the hold it in the ESPEasy repository, i will check it before implementing. :)
I tried to use ESPEasy numbers for the range 1-199 after that i've added some to the 200-300 range that is currently not implemented in ESPEasy but can be done someday, and if thought that a plugin will never be implemented in ESPEasy/ESP8266 - like USBRelay, VLC... - i've used range 500-600.
Controllers is a bit mixed, i am thinking about renumbering some new controller that is only exists in RPIEasy to range 100-200. (LoraDirect,BLE Direct, ESPNow receiver)

enesbcs, 2019-09-30:
> First of all, I want to say "wow" to the immense speed you manage to keep when building Rpi-Easy from scratch.
I've borrowed some HTML code from ESPEasy but yes, i worked a lot on it. It tooks maybe 6 months to get the first running version and announced it at 2019.01.01. 
I've learned a lot from the ESPEasy sources, altough i am not an expert Arduino/C programmer, the original structures turned to be very useful, except that the Raspberry Pi has a lot more RAM than the ESP8266, so i can hold almost every structures in the RAM in array that every device own properties can be found anytime. Adding a plugin in runtime was a great challenge, and i've never thought before that Python can be used in that way, but eventually it did.

Some problems take even more time, such as the self started AP, and the problem if the IR vs LIRC problem, but i have some ideas, the only thing that's never enough is time.

TD-er, 2019-09-30:
> Some problems take even more time, such as the self started AP, [...], but i have some ideas, the only thing that's never enough is time.

I totally agree :)
The instability we had has taken away so much dev time, so now all the ideas collected over the past 20 months can now get the attention they deserve.

enesbcs, 2020-04-08:
Closed as "In a separate file in the ESPeasy repository" accepted.

closed 2020-04-08 by enesbcs