enesbcs/rpieasy #95: Plugin P029 Output - Output Helper missing features

llevet, 2019-10-01:
Hi,
It miss on P029 the "interval" (report) like pluging P001.
"Send Boot state" will be welcome too.

Thank's.

Ludo.

enesbcs, 2019-10-01:
"Send boot state" and "interval" are both simply not applicable for output devices. It is not working in that way, neither in ESPEasy, nor RPIEasy.
Output helper is only for receiving states from the Controller, and it never returns anything to avoid infinite loops.

"Preserve state at startup" is a special setting which simply remembers output state and restore it at startup, this function is specifically requested by a user. (for example to HIGH if it was HIGH before reboot/halt)

llevet, 2019-10-01:
Hi,
No, you make a mistake.
On ESPEasy 'interval' for output send status of PIN (regardless of in or out) to the controler every interval.

A example of my jeedom log from a ESPEasy with a relay (interval set to 600s) : 
We've got a request for /device=SONOFF-P-R2&taskid=1&cmd=Relais2&value=0 from 172.31.31.134
Calling Jeedom http://127.0.0.1:80/plugins/espeasy/core/api/jeeEspeasy.php?apikey=b17b3jcmtmmFcaj6MqL3BOam3ynOLfXC&device=SONOFF-P-R2&taskid=1&cmd=Relais2&value=0&ip=172.31.31.134
Tue Oct 01 2019 15:33:56 GMT+0200 (CEST)Got response Value: 200
We've got a request for /device=SONOFF-P-R2&taskid=1&cmd=Relais2&value=0 from 172.31.31.134
Calling Jeedom http://127.0.0.1:80/plugins/espeasy/core/api/jeeEspeasy.php?apikey=b17b3jcmtmmFcaj6MqL3BOam3ynOLfXC&device=SONOFF-P-R2&taskid=1&cmd=Relais2&value=0&ip=172.31.31.134
Tue Oct 01 2019 15:43:56 GMT+0200 (CEST)Got response Value: 200

So i received the status of the relay every 10mn.
 - It permit to lost temporary connection beetween ESP and controler (frame lost), but the status is update after 10mn.
- If you reboot the controler or controler is offline for a moment (for updating it)  , it don't know the state of the different relays during the outage, but it will receive the status of all relays after 10mn in worst case.

'Send boot state' make the same (send status of the relay @ boot) 

Now, for INPUT pin, 'interval' is not always adapted because the controler will think that someone have push the button and is not the case and it can make some trouble on RULES on the controler.

 
Ludo.

llevet, 2019-10-01:
In fact, no need plugin at all.
According to  : https://www.letscontrolit.com/wiki/index.php/Relais#ESP_Easy
"Strictly seen, Relays are not a plug-in driven device like most other devices. It is not necessary to have a task (item under Devices tab) for them to work"

But every body use P001 to view and configure the PIN or catching events.
You can see it for example : 
https://lofurol.fr/joomla/electronique/domotique/160-espeasy-jeedom-piloter-un-relais-avec-une-wemos-lolin32-lite
Task 2 is a relais (P001).
or
https://itler.net/esp-easy-relais-ansteuern-mit-schaltzeiten/
or 
https://projetasgarddiy.fr/2018/06/14/interrupteur-sonoff-dans-domoticz/
GPIO-12 in the example.
or ....

In fact, on ESPeasy the P001 is AUTO CONF for input or output. But in RPIeasy , P001 switch is a true switch (only input)!  And P029 from ESPeasy is think for Domoticz MQTT Helper.

Your P029 is more generic  OUTPUT gpio.
So you understand why i wait for fonctionnality of P001 Espeasy on your P029.

My config of all my ESP@home for the Relays is  the same as top examples.
I use periodic feedback status to be shure that all is in sync (modules and Jeedom infrastructure).

Ludo.

llevet, 2019-10-01:
Found this ...
https://github.com/letscontrolit/ESPEasy/issues/2112
In fact it wasn't a desired normal operation but now it is adopted as is ...
So, it is why they haven't any OUTPUT RELAY plugin ...

But, i confirm that "Send Boot state" and "interval" are usable, and used (by me for example) for output pin.

Ludo.

enesbcs, 2019-10-01:
All right then please configure pin as output (Hardware->Pinout&Ports) and use P001 to report its state in RPIEasy also. It's working, I've just tested. Just make sure to enable "Force 10/sec periodic checking of pin", gpio command can be used to change its state.
P029 is meant to be one way, as it is designed to be used with Domoticz - but also can be used with any controller that did not want/accept reply (as a reply from an output device can lead to an infinite loop for Domoticz at least). As i am a Domoticz user, this is unlikely to change. :)

llevet, 2019-10-01:
Hi TD-er ,
Yes, you are right, a new plugin is need like OUTPUT P029 of RPIeasy.
But i'm only defends the presence of "Send Boot state" and "interval" on OUTPUT P029 that can be usefull in certain circumstances when use with controllers.

This is change the dialog direction : 
1) Controller can ask periodic status of all nodes
2) Nodes periodic inform the controller of there status 

With "Send Boot state" and "interval"  you have this 2 options. Without, only option 1.

I found solution 2 more elegant, controller have other things to do, and it doesn't need more code to retreive periodic status.

Ludo.

llevet, 2019-10-01:
Hi enesbcs,

As TD-er say, the best is to have a new plusgin (copy /paste of P001) with the good name and in OUTPUT mode whitout "Force 10/sec periodic checking of pin" workaround ...

Ok to live P029 as is, maybe change the name to reflect  Domoticz usage ...

Ludo.

llevet, 2019-10-01:
Hi enesbcs,

I have tested Output mode of P001. 
It seem there is a problem where we active "interval". The return value is change from integer to decimal : 
...
23:56:12: Event: test3#test3=0
23:56:23: Event: test3#test3=0
23:56:24: Event: test3#test3=0.0
23:56:34: Event: test3#test3=0.0
23:56:42: Event: test3#test3=1.0
...

But this problem is also appear with the previous version with INPUT mode.

Ludo.

enesbcs, 2019-10-02:
Issue addressed by commit https://github.com/enesbcs/rpieasy/commit/238fda597d4918e6ba8533a5cc508ba51158868b and commit https://github.com/enesbcs/rpieasy/commit/14875c558a4a5acc4ca8ba1ce97d4c07d6feb415

P001 issues: https://github.com/enesbcs/rpieasy/issues/96

closed 2019-10-02 by enesbcs