enesbcs/rpieasy #97: Add HTTP Status of GPIO

llevet, 2019-10-03:
Please add command (status) to return a Json state of GPIO.
(https://www.letscontrolit.com/wiki/index.php/ESPEasy_Command_Reference)

Example :
http://X.X.X.X/control?cmd=status,gpio,16

return :

{
"log": "",
"plugin": 1,
"pin": 16,
"mode": "input",
"state": 0
}

enesbcs, 2019-10-06:
Status command introduced in commit https://github.com/enesbcs/rpieasy/commit/0417591c885769b01160e45b2834257da3713598

llevet, 2019-10-06:
Nice.
can you add CR on each line ? Actualy is all inline, And replace ' by " like ESPeasy.

{
"log": "",
"plugin": 1,
"pin": 10,
"mode": "input",
"state": 0
}
instead of : 
{'plugin': 0, 'pin': 16, 'mode': 'input', 'state': -1}

Thanks.

llevet, 2019-10-06:
Also, it miss log.
When you send a GPIO command, the "log" is explicite like that :
wget http://172.31.31.138/control?cmd=GPIO,17,1 
{
"log": "GPIO 15 Set to 1",
"plugin": 1,
"pin": 15,
"mode": "output",
"state": 1
}

If it is a simple status, the log is "" : 
wget http://172.31.31.134/control?cmd=status,gpio,15
{
"log": "",
"plugin": 1,
"pin": 15,
"mode": "output",
"state": 0
}

enesbcs, 2019-10-06:
Sure i can replace single quotes to double, and adding CR is also possible.

I skipped "log" because i did not know what is the use of it.. and i still dont know what can i ever use it? If you issue a command, you know what you asked for.

enesbcs, 2019-10-07:
Status command output changed in commit https://github.com/enesbcs/rpieasy/commit/59c210ee89f96fbd638fe92f6697e9f1a0ce98e8

enesbcs, 2019-10-07:
Fixed in commit https://github.com/enesbcs/rpieasy/commit/3d8bf11e5d9f3220822fe9f65c4adeb0e8d1482a

llevet, 2019-10-07:
Perfect it's working.

closed 2019-10-07 by llevet