enesbcs/rpieasy #98: Add support for MCP23008 in plugin P009

llevet, 2019-10-06:
Hi,

Hi, thanks to add support for MCP23008 to th plungin _P009_MCP.

Ludo.

enesbcs, 2019-10-06:
Selectable type option added in commit https://github.com/enesbcs/rpieasy/commit/0417591c885769b01160e45b2834257da3713598

llevet, 2019-10-06:
Thanks, i will test it.
You can rename the plugin to : P009 - Extra IO - MCP230xx

enesbcs, 2019-10-06:
Only if works.. :D

llevet, 2019-10-07:
So,
Set output GPIO is ok :
http://172.31.31.138/control?cmd=MCPGPIO,2,0
http://172.31.31.138/control?cmd=MCPGPIO,2,1

If i put 1 GPIO to input with MCP interrupt to something, all other MCPGPIO change to the same MCP interrupt (input or output) and i can't remove it or replace it by a other pin.

Command to read status of MCPGPIO from http doesn't existe.

llevet, 2019-10-07:
[image]

enesbcs, 2019-10-07:
>If i put 1 GPIO to input with MCP interrupt to something, all other MCPGPIO change to the same >MCP interrupt (input or output) and i can't remove it or replace it by a other pin.
This is the way how the MCPGPIO driver works, it is using exactly ONE GPIO on the Raspberry to report changes from it's own pins.

> Command to read status of MCPGPIO from http doesn't existe.
This is why the first place the whole Controller architecture was involved... otherwise you can make your own rules to do anything you wish upon status changes like 
`on mpc-input-pin1#State do ... endon`

llevet, 2019-10-07:
1 - I agree that you have 1 GPIO (interupt line) for all the pins on the chip. But the problem is i can't change it on the fly. To change it, i need to delete all task and remake it...

2 - On my test, interupt line change only the status of input line. If i use this rule : 

on mpc-input-pin-3#state=1 do
   if [mpc-output-pin-2#state]=1
       MCPGPIO,2,0
   else
       MCPGPIO,2,1
   endif
endon

each time it go to MCPGPIO,2,1 branch. mpc-output-pin-2 state is 0 but in reality it is 1.

3 - And in final, calculation of adresss base (0x20 ..) doesn't work for MPC23008.
I put 2 MPC23008 and the interr
[image]
upt is the same (put automatically) for the two chip.
 
In fact, it will be very dificult for you to calculate the ship adresse of electronic component.
Imagine i use 1 MPC23008 and 1 MPC23017. 
So you can't calculate all possibility port: 
MPC23008 MPC23017 => 1-8 9-24
MPC23017 MPC23008 => 1-16 17-24
MPC23017 MPC23017 => 1-16 17-32
MPC23008 MPC23008 => 1-8 9-16
and this is only for 2 , think with more (max is 8) ...
Maybe the best is to put directly the address base on the interface like P023 with pulldown values : 0x20 0x21 .... 0x27 and not calculate it.
With that, it is easier to affect interrupt GPIO to each address base.

Sorry for all work i ask to you ....

enesbcs, 2019-10-07:
> 1 - I agree that you have 1 GPIO (interupt line) for all the pins on the chip. But the problem is i can't change it on the fly. To change it, i need to delete all task and remake it...

I do not see any reason why you can not change an interrupt - if you choose a correctly configured one (input,input-pullup, input-pulldown). The plugin tries to set it every time you press Submit.

> 2 - On my test, interupt line change only the status of input line. If i use this rule :
> 
> on mpc-input-pin-3#state=1 do
> if [mpc-output-pin-2#state]=1
> MCPGPIO,2,0
> else
> MCPGPIO,2,1
> endif
> endon
> 
> each time it go to MCPGPIO,2,1 branch. mpc-output-pin-2 state is 0 but in reality it is 1.

It is possible as outputs are not followed. I will look it up.

> 3 - And in final, calculation of adresss base (0x20 ..) doesn't work for MPC23008.
> I put 2 MPC23008 and the interrupt is the same (put automatically) for the two chip.
> 
> In fact, it will be very dificult for you to calculate the ship adresse of electronic component.
> Maybe the best is to put directly the address base on the interface like P023 with pulldown values : 0x20 0x21 .... 0x27 and not calculate it.
> With that, it is easier to affect interrupt GPIO to each address base.
> 
> Sorry for all work i ask to you ....

I am a little confused... you write an hour ago that you want to "stick to Espeasy behaviour"
https://www.letscontrolit.com/wiki/index.php/MCP23017
First extender starts from port 1, second starts from 17, third from 33 and so on. Regardless of its type. This is the current behaviour and the simplest one. If you use an MCP23008 you simply lose 8 ports, if you need these ports you have to buy an MCP23017. So MCP2 starts from 17 not from 9 as you thought. The interrupt will surely be the same for port 1-16 and a new interrupt can be assigned to 17-32.

llevet, 2019-10-07:
1 - I'm talking about changing : MCP interrupt. It is sticky to my first choice, can't change this GPIO. At validation, each time the old value is back.

3 - Nice ! I haven't think to that ! 
Add in source code documentation that in case of using MCP23008 the port count is the same as if we use a MCP23017.  

But the problem of MCP interrup is steel present. It put automatically the MCP interrupt of the first chip, no change is possible :
[image]

enesbcs, 2019-10-07:
In commit https://github.com/enesbcs/rpieasy/commit/3d8bf11e5d9f3220822fe9f65c4adeb0e8d1482a the GPIO changing is more easier, but only the first task can change it per device.
Now the plugin value state will follow output state. (At least I hope, because i have used my only one MCP for an ESP8266 long time ago, so i am unable to test it with an RPI.)

llevet, 2019-10-07:
Ok it is working now to change the MCP interrup and the second ship can have his own.

[image]

I will check the output state after (i need to push button but i can't make this remotely :-)

llevet, 2019-10-07:
Nice ! The follow output test is working ! My rule work without problem.

Thanks !

closed 2019-10-07 by llevet