enesbcs/rpieasy #100: Add new device EQ3 termostat over BT

tOOmki, 2019-10-08:
There are already projects on github, you just need to adapt it to RPIEasy.

enesbcs, 2019-10-08:
For python i found this project:
https://github.com/rytilahti/python-eq3bt
However integration of devices that i do not own, tends to be a little complex and takes several months...
But exactly what functions do you need from it?

tOOmki, 2019-10-09:
Sync:
 sync - Syncs time and prints target temperature and mode

Mode:
 auto - Sets auto mode and deactivates vacation mode if active.
 manual - Sets manual mode and deactivates vacation mode if active.

Temperature:
 temp <temp> - Sets target temperature to given value temp: 5.0 to 29.5 in intervals of 0.5°C, e.g. 19.5
 on - Sets thermostat to permanent on (30°C)
 off - Sets thermostat to permanent off (4.5°C)

enesbcs, 2019-10-11:
ESPEasy/RPIEasy can handle up to 4 variables per task. What variable do it needs? 1. target temperature (float), 2. mode (integer) , 3. temperature offset (float)...?
Mode can be: Closed, Open, Auto, Manual, Away, Boost. EQ3 manages 4,5 and 30 Celsius barriers on its own?
What do you mean under "prints target temperature and mod"? Reply with a json string to the sync command?

tOOmki, 2019-10-11:
> ESPEasy/RPIEasy can handle up to 4 variables per task. What variable do it needs? 1. target temperature (float), 2. mode (integer) , 3. temperature offset (float)...?
I dont now

> EQ3 manages 4,5 and 30 Celsius barriers on its own?

 **YES**
prints target temperature and mode = Shows the set temperature and the set mode

tOOmki, 2019-11-01:
How to use this commands ?

13:15:47 CMD: eq3sync
13:15:47 Unknown command: eq3sync
13:16:15 Event: Clock#Time=Fri,13:16
13:16:56 CMD: eq3mode,closed
13:16:56 Unknown command: eq3mode,closed

enesbcs, 2019-11-01:
Cool, is it working ok? Sync works, temperatures matches?

tOOmki, 2019-11-01:
Mode: -1.0** ???
19:46:43 CMD: eq3mode,closed
19:47:01 Event: Clock#Time=Fri,19:47
19:47:06 Event: GlowicaeQ-3#TargetTemp=-1.0
**19:47:06 Event: GlowicaeQ-3#Mode=-1.0**
19:47:08 CMD: eq3mode,closed
19:47:14 Event: GlowicaeQ-3#TargetTemp=4.5
19:47:14 Event: GlowicaeQ-3#Mode=0.0
19:47:14 Event: GlowicaeQ-3#TempOffset=1.0

Its correct ?
Mode=0.0 - close
Mode=1.0 - open
Mode=2.0 - auto
Mode=4.0 - manual

Its small problem with, temp below 20.5:
20:09:39 CMD: eq3temp,18.5
20:09:39 Unknown command: eq3temp,18.5
20:09:58 CMD: eq3temp,17.5
20:09:58 Unknown command: eq3temp,17.5
20:11:02 CMD: eq3temp,19.5
20:11:02 Unknown command: eq3temp,19.5
20:11:08 Event: Clock#Time=Fri,20:11
20:11:17 Event: GlowicaeQ-3#TargetTemp=19.5
20:11:17 Event: GlowicaeQ-3#Mode=3.0
20:11:17 Event: GlowicaeQ-3#TempOffset=1.0
20:11:32 Event: GlowicaeQ-3#TargetTemp=19.5
20:11:32 Event: GlowicaeQ-3#Mode=3.0
20:11:32 Event: GlowicaeQ-3#TempOffset=1.0
20:11:33 CMD: eq3temp,20.5
20:11:45 Event: GlowicaeQ-3#TargetTemp=20.5
20:11:45 Event: GlowicaeQ-3#Mode=3.0
20:11:45 Event: GlowicaeQ-3#TempOffset=1.0

enesbcs, 2019-11-01:
As i sad originally "However integration of devices that i do not own, tends to be a little complex" this is why i usually dislike this method, as i can not see what is happening in the background, i can only guess it.

The modes based on the library:
https://github.com/rytilahti/python-eq3bt/blob/master/eq3bt/eq3btsmart.py
    """ Thermostat modes. """
    Unknown = -1
    Closed = 0
    Open = 1
    Auto = 2
    Manual = 3
    Away = 4
    Boost = 5

I guess "-1" used when there are no known state/no communication with the device or the status update package not yet arrived from it. I recommend to place somewhere nearer posititon to the Raspberry Pi or buy a more powerful USB BLE stick into it.

"Unknown command:" is the standard answer if command is not succesful, for example: connection can not be established with the device. I can modify it for you to write "Connection failed" but it doesn't really matter, the result will be the same: command execution failed.

Hints: 
-Most BLE device rejects more than one connection, so if you connected with your phone, than the RPI cant connect with it to the same time.
-BLE connection is not so fast, may take 3-5 seconds, if you try to execute commands at a faster rate it will fail

enesbcs, 2019-11-06:
As i can not achieve more with this plugin without a device, the issue is now closed.

closed 2019-11-06 by enesbcs

enesbcs, 2020-03-02:
Be warned that command structure is changed now as per issue https://github.com/enesbcs/rpieasy/issues/132