enesbcs/rpieasy #102: AP mode not working

pczolee, 2019-10-31:
Hello.

Unfortunately for me the AP mode not working, but this is the smallest problem, the biggest is, if I want to use my own config (systemwide), I cannot disable the rpieasy ap management, so when the rpieasy starting it stopping hostapd and dnsmasq, and deleting (!) -without backup- dnsmasq.conf.

Please, make a switch to disable wlan ap management from web, (or make it, when the "Start AP when:" is "Never") and if it is disabled let the system run ap settings as it is configured systemwide!

Otherwise. I think the best way is to use your config files, and run the dnsmasq and hostapd with these (so stop the services -if they are running, and ap mode enabled from your web- and than run the programs with an argument and use your config file), and do not touch system config!

Thanks

enesbcs, 2019-10-31:
I see. I saw a solution which used custom config file, but running the program in foreground blocks further execution, so it has to be reworked for multithreading in python code if changed that way.
Maybe the easiest solution is that "AP starting Never" mode will be the new default, which will leaves the system config untouched.

enesbcs, 2019-10-31:
Done. Now the "Never" option is the default on new installations, if "Never" setting is active none of the system conf file touched.

pczolee, 2019-11-04:
Thanks. Now it looks good (at least the system files not touched). 

Otherwise, it is using the lib/scripts/ap_start.sh / ap_stop.sh ? Because the other problem was that the old config was just deleted, did not make a backup. But the script above should make a copy (line 50: sudo cp /etc/dnsmasq.conf /etc/dnsmasq.conf.backup), so if this is running, than why not made copy?

enesbcs, 2019-11-04:
It's because i left somehow an extra "rm /etc/dnsmasq.hosts" line in the end of the ap_stop.sh, and i just fixed it at 4 days ago, after you pointing to this error. 
I hope it is now working as planned.

enesbcs, 2019-11-06:
As no negative feedbacks arrived i think it is solved for now.

closed 2019-11-06 by enesbcs

happytm, 2019-11-09:
I came across this. Just linking for your reference:

https://github.com/Phoenix1747/RouteryPi

Thanks

enesbcs, 2019-11-09:
Thanks, but AP mode is working according my tests. This 102 issue is not about "not working" but instead "AP mode working and deleting custom AP mode".