enesbcs/rpieasy #104: Docker image for RPIEasy

happytm, 2019-10-31:
I have reommended RPIEasy for docker image. This is just for your information.

https://github.com/gcgarner/IOTstack/issues/10

Thanks.

enesbcs, 2019-11-09:
https://github.com/gcgarner/IOTstack/issues/26

gcgarner, 2019-11-28:
Is there an easy way to run RPIEasy in the background or on startup?

I tried:
`sudo RPIEasy.py &` and it still outputs to the terminal. Should i add a `> /dev/null` redirect?

When running this native is there any easy way to change the default port? 8080 is quite popular

enesbcs, 2019-11-28:
Yes there are a super easy way: if it is started by root or sudo, then at the Hardware menu at the webgui there are an option for "RPIEasy autostart at boot" (try to use "screen" if installed). Although you can try with & > /dev/null but "screen" is a better way.

Port is autodetected at startup, (80,8080,8008) which is free it will grab it.

happytm, 2019-12-04:
I think it is implemented now.

https://github.com/gcgarner/IOTstack/issues/26#issuecomment-559644600

Thanks.

enesbcs, 2019-12-05:
Cool! Did you test it?

happytm, 2019-12-05:
No. I am away from my PI Zero right now at least for a week . 

Thanks.

enesbcs, 2020-03-13:
Closed as resolved.

closed 2020-03-13 by enesbcs