enesbcs/rpieasy #109: neopixel snd neopixelline issues

scargill, 2019-12-14:
I managed to get neopixelall to work, but neopixel is unrecognised and neopixelline - first use - say starting at 2 and ending at pixel 4 works but subsequent changes don't. Is this only partly implemented or am I missing something? 

I have a 12-LED strip as a test.

Pete

enesbcs, 2019-12-15:
> I managed to get neopixelall to work, but neopixel is unrecognised and neopixelline - first use - say starting at 2 and ending at pixel 4 works but subsequent changes don't. Is this only partly implemented or am I missing something?

This three commands are implemented in Neopixel plugin:
NeoPixel,[led nr],[red 0-255],[green 0-255],[blue 0-255],[brightness 0-255]
NeoPixelAll,[red 0-255],[green 0-255],[blue 0-255],[brightness 0-255]
NeoPixelLine,[start led nr],[stop led nr],[red 0-255],[green 0-255],[blue 0-255],[brightness 0-255]

In NeoPixel command there are only two condition that makes it silently refuse to work: 1. HPWM compatible pin is not selected in the plugin, 2. LED number in command greater or equal than setted LED count number. (led number are zero based) If NeoPixelAll works for all 12 LEDs than maybe some other error occurs.
On other errors RPIEasy writes them on the error log, do you see anything in the error log? Please add the content of the Log and the exact command lines in case of error.

scargill, 2019-12-15:
Sure enough, I got neopixel wrong - I forgot to put in the brilliance - 

So..
http://192.168.14.70/control?cmd=NeoPixelall,255,0,0,20
Lights up all 12 LEDs a dim RED as I'd expect. Then:
http://192.168.14.70/control?cmd=NeoPixelall,0,0,255,255
changes all 12 to bright blue.

The above works a treat..
This
http://192.168.14.70/control?cmd=NeoPixel,3,255,0,0,255
leaves them all on blue but the THIRD one which it light up RED.
All correct..

BUT THIS... changes ONLY LEDS 3 and 4 in GREEN leaving the rest as they were (in DEVICES for Neopixel I have 12 LEDS) only... it looks like it SHOULD light up from 3 to 7.
http://192.168.14.70/control?cmd=NeoPixelline,3,7,0,255,0,255

That line above generates "true" as do the others - and I have NO idea where any log file for this is? Am I missing something?

TD-er, 2019-12-15:
> BUT THIS... changes ONLY LEDS 3 and 4 in GREEN leaving the rest as they were (in DEVICES for Neopixel I have 12 LEDS) only... it looks like it SHOULD light up from 3 to 7.
> http://192.168.14.70/control?cmd=NeoPixelline,3,7,0,255,0,255

The other commands you sent were using 5 parameters. This one has 6.
Maybe a typo?

scargill, 2019-12-15:
Hi TD-ER
According to ENESBCS:
neopixelall has 4 params
neopixel has 5 params
nepixeline has 6 params

and thats what I used above - and all now work except neopixelline

scargill, 2019-12-15:
http://192.168.14.70/control?cmd=NeoPixelline,2,8,0,0,255,20

No doubt in my mine - that second parameter (8)- the last led to colour - is not doing anything.

TD-er, 2019-12-15:
Maybe @enesbcs can shed some light on this then :)
I guess the code may differ from what we use in ESPEasy, as the comment there states:

```c++
//#######################################################################################################
//#################################### Plugin 038: NeoPixel Basic #######################################
//#######################################################################################################

// 15-June-2017: Fixed broken plugin; tested with several neopixels (single LED, 8 LED bars and 300 leds strips.

// List of commands:
// (1) NeoPixel,<led nr>,<red 0-255>,<green 0-255>,<blue 0-255>
// (2) NeoPixelAll,<red 0-255>,<green 0-255>,<blue 0-255>
// (3) NeoPixelLine,<start led nr>,<stop led nr>,<red 0-255>,<green 0-255>,<blue 0-255>
```

It seems like we also need to add the last "brightness" parameter for ESPEasy as that sounds like a useful parameter.

enesbcs, 2019-12-15:
@TD-er: In fact, the last parameter "brightness" is just optional, if middinh RPIEasy simply using the default brightness that is setted on the plugin page. :)
Thanx for testing scargill, i have a 8 LED array for testing, i will try neopixelline,2,4,... then neopixelline,2,8,...
Console logs are fine, but you can raise log level to Debug in Tools/Advanced menu.

enesbcs, 2019-12-15:
I've found an inversed condition in the led number checking logic, i think it is now OK in commit https://github.com/enesbcs/rpieasy/commit/4ca47e520f957c8c21f91bb029fe2bab0e91cdce

scargill, 2019-12-15:
Sorry I was going to check into further debugging but the broadband went out and I’ve spent the entire night complaining to utterly useless Vodaphone in the UK.

 

I will check when I figure out how to update to that commit. Marvellous….

 

Regards

 

Pete

scargill, 2019-12-15:
Yup, I just grabbed the neopixel file, rebooted and it works wonderfully. Thank you!!

enesbcs, 2019-12-16:
Thank you for you feedback! Issue is now close, feel free to reopen in case of connected bug.

closed 2019-12-16 by enesbcs

scargill, 2019-12-16:
Not a bug but I cant think where eles to contact you... rpieasy - all woprking but of course dies when I close the ssh session to the pi. How would I have it run like a service at Pi power up without terminal interaction. Can you suggest anything. I'd like to blog about thie project.  Regards, Pete

enesbcs, 2019-12-16:
https://github.com/enesbcs/rpieasy/issues/106